valterc/ki2 #29: Use hood-button double-click to zoom map

kintrupf, 2022-11-03:
Zooming in and out of the map with long-holding the hood-buttons is very imprecise. There is a very noticeable lag between the button press and the reaction of the map. That's obviously not a ki2 problem, this was also a present in the original Karoo di2 support.

I think mapping the double-click left and right to zoom-in and zoom-out would work much better.
Could you add this options to the double-click actions?

valterc, 2022-11-04:
Good suggestion, I noticed the same. The problem comes from di2 itself, the hood button event is delayed for a second or so.

valterc, 2022-11-13:
This is now possible in release [0.5](https://github.com/valterc/ki2/releases/tag/0.5), you can map _single press_ or _double press_ to zoom out/in the graph or map. Note that this only works if the map or graph is visible in the current selected Karoo page.

[image]

closed 2022-11-13 by valterc