valterc/ki2 #30: Left and right switched?

kintrupf, 2022-11-04:
Just a very minor issue, I noticed that when I select "Navigate left" for a hood button the Karoo acts as if I had pressed the *right* side button and vice versa.

Same for the "zoom in" and "zoom out" functions. If I set the left button long press to "zoom out" and "zoom in" to the right button long press (as it was on the original Karoo implementation) it would actually zoom in on left and zoom out on right 😆

These issues were quickly fixed during my first ride, just opened the ki2 app and changed the button actions on the fly 😉

valterc, 2022-11-06:
I was unable to reproduce this issue. I suspect the issue might be caused by your di2 configuration.

The way the hood buttons work is they emit "Channel" events, the Ki2 app expects Channel 1 for Left and Channel 2 for Right. These channels are configurable in E-Tube:
[image]

If your configuration is different than Channel 1 for Left and Channel 2 for Right (like that screenshot in E-Tube) then it won't translate well in Ki2 because that's what it expects.

It could be some other problem but I wasn't able to find anything else. 

Just to confirm:
- Are you using Karoo 1 or Karoo 2? 
- What do you see in the Ki2 app, tap on the di2 device, wait for it to connect and then see what is displayed in the Left / Right Switch when you press the hood buttons - is it also flipped there?
  
[image]

Thanks.

kintrupf, 2022-11-08:
You're absolutely right! I had my hood buttons set to shift up and down after the original shifter support from the Karoo was removed. After installing ki2 I found that the hood buttons weren't working as expected and played a bit with the settings in the eTube app, and that resulted in Channel 1 being assigned to the right button and Channel 2 for the left one 🙄

You should write something about the expected channels in the README.md 😉

valterc, 2022-11-08:
Thanks for checking. To be honest I didn't even consider this before, I will add it to the Readme 🙂