valterc/ki2 #39: Conecting to Device

s3dbw, 2022-11-21:
Sideloaded onto the Karoo 2 no problem, however I am having problems connecting to the DI2. The app starts the search but cant seem to find the DI2 on the bike? Any suggestions?

JayUK, 2022-11-21:
Is your Karoo running the correct version of Hammerhead software (ie: one that no longer supports Di2)?

valterc, 2022-11-21:
I've created a small video, make sure to follow the steps (enable sound):

https://user-images.githubusercontent.com/1299179/203050478-5d664a74-ca16-428b-b3f3-21ef1ec9d5df.mp4

- Make sure you don't have DI2 paired with Karoo (if you are on an old Karoo software version that still supports it)
- Make sure you have less than 14 ANT devices added/paired in Karoo
- Open Ki2 and press the + button, searching for DI2 devices will start
- On your Di2 system, you need to enable connection/pairing mode. This is usually done by holding the Di2 function button for a second or so. (The green and red lights in Di2 will flash in alternating order - if your version has those.)
- The Di2 device should appear in Ki2 in the search results list.

Let us know if that works for you 🙂

closed 2022-12-02 by valterc