valterc/ki2 #44: Installing V0.7

s3dbw, 2022-12-02:
Hi there, just tried installing the latest version, however getting the following error message. Does V0.6 have to be uninstalled first?

Thanks

[image]

valterc, 2022-12-02:
(You can uninstall before, in that case you will lose the added Di2 device and settings, instead) you should run the command with the **-r** argument, like so:
`adb.exe install -r "apps\ki2-0.7-release.apk"`

Let me know if that works. I'm working on auto updates for the next version :)

s3dbw, 2022-12-02:
That did the trick. Auto updates would be brilliant. Thanks for all your efforts

closed 2022-12-02 by valterc