valterc/ki2 #46: Reddit traction

spencertollefson, 2022-12-03:
Not an issue, but wanted to let you know that I think you may have gotten some new traction today from a reddit thread. Thanks for making and sharing this project!

reddit.com/r/Karoo/comments/zadz78/ki2_app_for_karoo_to_restore_di2_functionality/

byronbrewer, 2022-12-03:
That's how I landed here.

Thank you so much for your efforts!

valterc, 2022-12-03:
Thanks folks 🙂 I saw the posts on reddit and facebook. I wanted to keep a low profile until the app is a bit more solid and tested, but I guess the beans are spilled now 😁 

(Hopefully HH keeps supporting the Karoo SDK)

Ruud-C, 2022-12-03:
Great stuff, thank you! Your work is much appreciated.

valterc, 2022-12-09:
Enjoy guys. If you have problems or suggestions, feel free to reach out!

closed 2022-12-09 by valterc