valterc/ki2 #47: Add a switch to Map switch

judahrand, 2022-12-03:


valterc, 2022-12-04:
Just to clarify: press the hood buttons to switch to the map page in Karoo? 

That is currently not possible with the way Karoo works. Each action (like navigate page left, navigate page right, zoom map, etc.) is bound to a keyboard event. The hardware buttons emit these keyboard events, Ki2 is doing the same, and the Karoo app executes them. There is no such event to 'switch to map page'. 

It might be possible to do with some trickery but it's unclear if we can make it 'stable' enough to avoid it breaking with each Karoo software release.

valterc, 2022-12-18:
This will be available in the next version. Still need to do some more testing.

closed 2022-12-18 by valterc

judahrand, 2022-12-18:
> This will be available in the next version. Still need to do some more testing.

Incredible! This has project and especially this change (+ self update) has made me want to use my Karoo 2 again after months! Thank you.

valterc, 2022-12-18:
That is nice to hear 🙂 thank you! The update is available, you still need to manually update this time.

- Download the APK file from [here](https://github.com/valterc/ki2/releases/tag/0.8)
- Run the install command with the **-r** parameter like so: `adb.exe install -r "APK-FILE-PATH"` to avoid needing to uninstall the previous Ki2 version
- After updating, restart the Karoo before riding

PR #65 (2022-12-18): Add 'switch to map page' button option
Resolves #47