valterc/ki2 #49: Functionality dies if there is a device data field without a device

freedman14, 2022-12-09:
- Inserted a graphical heart rate data field
- Inserted a numeric gear field on the same p age
- Did **not** have a heart rate monitored paired to device
- After @ 15 mins of riding, the shift indicator started to become unresponsive
- After @ 5 additional minutes, shifter data field displayed N/A and become non functional

valterc, 2022-12-09:
Interesting... I know that Karoo restarts the ANT service on some occasions, maybe that could be happening there. Will need to test. 

Did this happened once or it always happens?

freedman14, 2022-12-09:
Once I removed the heart rate data field, worked perfectly.  Rode for an hour + and no issues.

valterc, 2022-12-16:
I wasn't able to fully replicate this issue, but I got to simulate it and the problem was there indeed. I've made some improvements, will still do some more testing - the fix will be available in the next release.

closed 2022-12-16 by valterc

freedman14, 2022-12-16:
Thanks!  I can do a test when the next release comes out.

PR #62 (2022-12-16): Improve ant reliability
Fixes #49