valterc/ki2 #50: Pause / Lap on long press

akivanwasser, 2022-12-09:
Is it possible to add the ability to map a long press on the hood buttons to one of the bottom buttons on the K2 so it can be used for pausing and / or lapping.

This way it would be possible to swap from the original behavior of the buttons.

valterc, 2022-12-09:
Should be possible, thanks for the suggestion.

valterc, 2022-12-14:
Will be available in the next release.

closed 2022-12-14 by valterc

valterc, 2022-12-18:
The update is available, you still need to manually update this time.

- Download the APK file from [here](https://github.com/valterc/ki2/releases/tag/0.8)
- Run the install command with the **-r** parameter like so: `adb.exe install -r "APK-FILE-PATH"` to avoid needing to uninstall the previous Ki2 version
- After updating, restart the Karoo before riding

PR #58 (2022-12-14): Add option to pause/resume or mark lap on button hold
Resolves #50