valterc/ki2 #51: ki2 crashing with Specialized Turbo Creo di2

robboyer, 2022-12-09:
Hello

I successfully paired ki2 to my Turbo Creo's di2, but when I start the ride the ki2 app immediately crashes, 100% repro.  I'm thinking perhaps this is because the bike has no front derailleur?  (it's a gravel bike)

Everything works fine with my Specialized Aethos di2 setup (it does have a front derailleur)

Anyway, it would be great if the issue could be fixed.

Please let me know if I can help debug the issue.

Thanks
-Rob Boyer

robboyer, 2022-12-09:
forgot to mention, THANKS FOR CREATING THIS AWESOME APP!  I was really missing the di2 integration.

robboyer, 2022-12-09:
I pulled the top of the crash stack off the device.  It does appear to be trying to update the front derailleur graphic:

12-09 08:39:55.772  2992  2992 E AndroidRuntime: FATAL EXCEPTION: main
12-09 08:39:55.772  2992  2992 E AndroidRuntime: Process: io.hammerhead.rideapp:io.hammerhead.rideapp.rideActivityProcess, PID: 2992
12-09 08:39:55.772  2992  2992 E AndroidRuntime: java.lang.IllegalArgumentException
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.graphics.LinearGradient.nativeCreate2(Native Method)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.graphics.LinearGradient.createNativeInstance(LinearGradient.java:108)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.graphics.Shader.getNativeInstance(Shader.java:160)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.graphics.Paint.getNativeInstance(Paint.java:649)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.graphics.BaseCanvas.drawRoundRect(BaseCanvas.java:362)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.graphics.Canvas.drawRoundRect(Canvas.java:1793)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at com.valterc.ki2.views.GearsView.drawFrontGears(GearsView.java:229)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at com.valterc.ki2.views.GearsView.drawPicture(GearsView.java:182)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at com.valterc.ki2.views.GearsView.onLayout(GearsView.java:138)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.view.View.layout(View.java:19663)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at androidx.constraintlayout.widget.ConstraintLayout.onLayout(ConstraintLayout.java:71)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.view.View.layout(View.java:19663)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.view.ViewGroup.layout(ViewGroup.java:6075)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.widget.RelativeLayout.onLayout(RelativeLayout.java:1083)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.view.View.layout(View.java:19663)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.view.ViewGroup.layout(ViewGroup.java:6075)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at androidx.constraintlayout.widget.ConstraintLayout.onLayout(ConstraintLayout.java:71)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.view.View.layout(View.java:19663)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at android.view.ViewGroup.layout(ViewGroup.java:6075)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at androidx.recyclerview.widget.RecyclerView$LayoutManager.layoutDecorated(RecyclerView.java:21)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at io.hammerhead.dataelements.views.adapter.SpannedGridLayoutManager.hhd(SpannedGridLayoutManager.java:182)
12-09 08:39:55.772  2992  2992 E AndroidRuntime: 	at io.hammerhead.dataelements.views.adapter.SpannedGridLayoutManager.onLayoutChildren(SpannedGridLayoutManager.java:414)

valterc, 2022-12-09:
Likely because of no front/rear derailleur indeed. Thank you for the stack, will help to fix 🙂 I'm working on some other things, I should be able to fix it this weekend.

robboyer, 2022-12-09:
excellent, thank you

valterc, 2022-12-14:
This is now fixed and will be available soon in a new release. Still need to go out for a ride to test :)

closed 2022-12-14 by valterc

robboyer, 2022-12-14:
Excellent!  I’m available to test any pre-release, if it would be helpful.

Best,
-Rob Boyer

valterc, 2022-12-18:
The update is available, you still need to manually update this time.

- Download the APK file from [here](https://github.com/valterc/ki2/releases/tag/0.8)
- Run the install command with the **-r** parameter like so: `adb.exe install -r "APK-FILE-PATH"` to avoid needing to uninstall the previous Ki2 version
- After updating, restart the Karoo before riding

robboyer, 2022-12-21:
Just tried 0.8, problem is fixed.  No issues the entire ride on my Specialized Creo (1x11) gravel bike.  Thanks!

PR #56 (2022-12-14): Fix crash when using single/fixed gear bike
Fixes #51