valterc/ki2 #64: Gear ratio

josesoriano, 2022-12-16:
I don't know how are this would be but I would throw it out there.

Is it possible to add and show the gear ratio instead of numbers? for me, numbers are nothing, gear ratio is huge.

I donated for your coffee. I hope it's not Starbucks xD 

I will keep donating to maintain and improve it. Thank you!!!!

valterc, 2022-12-16:
Many thanks for the coffee 🙂 and for the suggestion. 

I agree that's a nice thing to have, but it's going to take a while, it's a big thing to add.

josesoriano, 2022-12-17:
What if, you just allow people to label the current numbers with their gear ratio, you don’t need to detect anything, it’s just a label Sent from my iPhone

valterc, 2022-12-28:
> What if, you just allow people to label the current numbers with their gear ratio, you don’t need to detect anything, it’s just a label

Totally agree, the problem is that it must be configurable per each Di2 device, so I would need to rework quite a lot of things to support that. Currently all configuration is global.

Nonetheless I've been looking into this and I think we can get the chainring information from the shifter, so we could eventually display the gear size (teeth count) and the calculated ratio between front and rear.

josesoriano, 2022-12-28:
That’s amazing!!You would be a rock star for a lot of us if you achieve thisSent from my iPhone

ts-av, 2022-12-31:
I just pulled code to try and add that. From what I've seen, getting the information from the Di2 is inconsistent (when I tried to get it on my wife's Garmin it didn't work) so I was going to just have a list of standard setups for now. For example :

**R9200/R8100 (12s) 11-30** | 11 | 12 | 13 | 14 | 15 | 16 | 17 | 19 | 21 | 24 | 27 | 30
**RX810 48/31**|48|31
 It does complicate things for people with more than one bike and different configurations. I wonder if there's a way to utilize profile based settings to do that (not sure HH added that to the API yet, I'll have a look)

josesoriano, 2022-12-31:
That’s amazing man! Sent from my iPhone

valterc, 2022-12-31:
> getting the information from the Di2 is inconsistent (when I tried to get it on my wife's Garmin it didn't work even though it has an auto-detect option)

This is because the information for Di2 is not the actual individual chainrings, so you won't get something like `11-30 : 11 | 12 | 13 | 14 | 15 | 16 | 17 | 19 | 21 | 24 | 27 | 30`, instead you get an identifier, like the number `5`. So it's the responsibility of the head unit to recognize the identifiers and use the correct chainring information. For this to work both the Di2 needs to be configured correctly via E-Tube and the headunit/app needs to be aware of all known identifiers. If new chainrings come to market, headunits need to be updated or if you have a custom chainring then it might not be detected correctly 😐 

I've started working on here: https://github.com/valterc/ki2/tree/gear-ratio / https://github.com/valterc/ki2/commit/425853ac2a83bab577cbea552d379b0446a9c5b4

> It does complicate things for people with more than one bike and different configurations. I wonder if there's a way to utilize profile based settings to do that (not sure HH added that to the API yet, I'll have a look). Alternative is adding an array in the global setting with unit ID as first element, followed by gears so that the program loads the correct gears on initial connection.

Yeah this is the main problem, that's why I won't be adding a custom way to set the gears for now. There is no API for the profiles in the Karoo SDK. Adding an array with the Id won't scale well once we start to have more configurations per unit, especially in the UI part.

ts-av, 2022-12-31:
> I've started working on here: https://github.com/valterc/ki2/tree/gear-ratio / [425853a](https://github.com/valterc/ki2/commit/425853ac2a83bab577cbea552d379b0446a9c5b4)

Thanks, I'll pull that branch instead. Also, attached is a list of all the current sets I know of. Unfortunately I don't know which one corresponds to which ID from the Di2 unit. 
[gear ratios.csv](https://github.com/valterc/ki2/files/10327618/gear.ratios.csv)

valterc, 2023-01-03:
This will be available in the next version. A new data element will be available with the current Gear Ratio. You need to make sure the Di2 is correctly configured (from E-Tube) with the correct chairing settings. I will add an example of what you need to set later.

closed 2023-01-03 by valterc

josesoriano, 2023-01-03:
Pretty amazing!

As soon as you release it, I will push people on the fb group to donate
man! I appreciate your hard work

[image: facebook] <https://www.facebook.com/josesorianophotography/>
[image: linkedin] <https://www.linkedin.com/in/josesorianocovello/>
[image: instagram] <https://www.instagram.com/josesorianophoto/>
Jose Soriano

Visual Content Creator

Jose Soriano Photography
4034002544
***@***.***
www.josesoriano.com

josesoriano, 2023-01-13:
> This is now available in the latest release - you can update directly from Ki2.
>
> You can add the gear ratio element to Karoo ride profiles from the Karoo profile editor.
>
> For the gear size (teeth count) and gear ratio to work correctly the chainring information must be configured in Di2 via the E-Tube app (should be already correct but double check if not working as expected):
>
>  <https://user-images.githubusercontent.com/1299179/212357836-e01a0b5a-8fd2-4a6b-8803-7a4b670792b7.png>

This is great. 

Do you want to make the release on fb group? I can push for people to donate.

This is a great update

Jose Soriano
Visual Content Creator
Jose Soriano Photography

4034002544
***@***.***
www.josesoriano.com

PR #81 (2023-01-03): Add gear ratio
Resolves #64