valterc/ki2 #69: Numeric view goes to "N/A" on longer rides

freedman14, 2022-12-18:
1. Ride was 3+ hours
2. Two additional Ant devices connected  > Stages Power Meter And Heart Rate Monitor
3.  Shifting display worked great until @ 2:45
4. At that point, gear numbers went to N/A
5. Device was still connected -- could still use the Di2 Hood buttons to switch screens

valterc, 2022-12-19:
I assume you were using version 0.7?

This will be tricky to debug, interesting that the switches continued to work when the displays did not. Do you have a single Di2/bike added to the app or multiple ones?

freedman14, 2022-12-19:
1. 0.7
2. Single bike
3. I upgraded to 0.8 and will see if I can replicate

robboyer, 2022-12-19:
Hi Valter,

I’ve been having a similar issue - the “Gears” field changes to “Waiting for Data…” (not “N/A”)  and sometimes re-appears later, sometimes not.  Strangely, the hood switches continue to work.

Since I have 2 bikes with di2, I did have two Di2 setups in ki2 app.  So yesterday when the problem happened at the beginning of the ride I ended the ride,  deleted the di2 setup not being used from the ki2 app, and started a new ride.  The gear display worked the entire 40 mile ride after that without incident.  

Is there anything I should look at if the problem re-appears that would help you debug it?

Will also try 0.8 this week.

Thanks for working on this project!

-Rob

s3dbw, 2022-12-19:
This might actually be an issue with the Karoo software? Reason I say this was that before HH dropped DI2 support, I occasionally had a similar issue where it would momentarily loose the DI2 gear indication, then pick up again. Strangely enough it happened most often when changing between gear 10 and 11? Dont know if there was any significance in that or not?

regards

freedman14, 2022-12-19:
@valterc  Let me know if attaching the .fit file would help.

valterc, 2022-12-23:
Thanks freedman14, I don't think the FIT file would help and may contain private info like the location, please don't attach it here! 🙂 

It's winter time on my location so I don't really go for long rides, I've tested the app at home and it stayed connected for over 10 hours. This was with last Karoo update and Ki2 version 0.8 and some other minor chances that I made in #72 (this will also fix some issues when having multiple Di2 devices added).

[image]

If you are still seeing issues we need further debugging... One other user also reported problems when using workouts, I haven't looked into that one yet.

freedman14, 2022-12-23:
Here is my bizzaro theory.  I think there is an issue with the head unit when you ride below 40 degrees F.  Would be interested in if others are affected by inconsistent ant functionality in the winter.

Nimmerichter, 2022-12-25:
Hi all,
to add to this topic:
in my case, the fields changed to N/A after 10-15min. Interestingly though, shift count works, as does the hood buttons. However, gears and battery status does not. I have also checked the Ki2 app and the status is connected.
I am on version 0.7 and have two Di2 bikes added (obviously not connected at the same time).
cheers
alfred

valterc, 2023-01-02:
I've made further improvements to the ANT connection, it should help with connection stability on long rides. I've also improved the way the app handles several Di2 units (even if only one is connected). Both these changes should help with the connection issues.

Please update to [version 0.9](https://github.com/valterc/ki2/releases/tag/0.9) and give it a go when you can. If you have version 0.8 you can update directly from the Karoo, the app should show a notification for auto-update or you can trigger an update via the settings.

freedman14, 2023-01-03:
Thanks @valterc .  Loaded up .9 and rode it yesterday for @ 2 hours.  Few observations

1. Seemed to pickup the gearing faster on boot up
2. It did one N/A during an auto-stop but picked up the gearing once I started riding again
3. The permission screen is a bit wonky to proceed get out of.  Not sure if there is anything you can do about this as I'm assuming it's either a K2 or Android screen

Great job.  Thanks so much for your great work!

JNF

Nimmerichter, 2023-01-03:
Thank you Valter,
haven't had the opportunity to ride outside due to poor weather conditions. However, update from 0.8 to 0.9 works seamlessly and a short 90min indoor ride today revealed no issues.
Hope I can provide further informations after some outdoor rides.

Great work, much appreciated
cheers
alfred

Nimmerichter, 2023-01-07:
Re to my last post, I did a 3h ride today at 8°C and had no issues with 0.9.

many thanks
alfred

jeremywj, 2023-01-12:
> Here is my bizzaro theory. I think there is an issue with the head unit when you ride below 40 degrees F. Would be interested in if others are affected by inconsistent ant functionality in the winter.

I have been riding consistently in 20-45F degree weather (typically lower 30s) and have not had this issue.  I did a 70 mile ride in 35F a week or so ago and it never dropped once.

valterc, 2023-02-21:
I made some fixes in the latest version [2.0](https://github.com/valterc/ki2/releases/tag/2.0). Please install/update.

closed 2023-02-21 by valterc