valterc/ki2 #70: Feature request: beep when switching to highest or lowest gear

robboyer, 2022-12-21:
One of my favorite things about the Garmin computers' support of di2 is that they beep (pretty loudly) when you shift to either the highest or lowest gear.  This is really useful to hear in cases where you're on a screen not showing the gears, or you can't safely look at the screen (for example, while going fast down a curvy road).  It's a great way to know that you "have no more gears" 

Not sure it's possible given the API you have to work with, but it would be awesome if ki2 could do this. 

Thanks for considering.

valterc, 2022-12-22:
That is a nice suggestion! I need to understand what is the best to do that, but should be possible, thanks.

kemaMartinsson, 2022-12-22:
Suggestion. Make it configurable in Ki2.
The ones using the display eg MT800-system display already have that beep.  
It would be annoying to have multiple beeps.

robboyer, 2022-12-22:
@kemaMartinsson I don't think the high/low gear beep is configurable on Garmin computers, it's always on.  What is an MT800-system?

kemaMartinsson, 2022-12-22:
@robboyer I thought this was about Ki2 running on Karoo not Garmin, my suggestion was to make it configurable in Ki2.  
[SC-MT800](https://www.bike24.com/p2172970.html) is   the display for XT Di2 and it beeps on high/low gear (if configured in E-Tube).

robboyer, 2022-12-22:
@kemaMartinsson thanks for the MT800 info.   I was talking about the Garmin computers like the Edge series -  just using them as an example for the Karoo+ki2 implementation.  I wasn't aware you could turn on/off the beeping in E-Tube, that's good info.  Although hopefully I'll never have to go back to the Edge now that we have ki2 😄

valterc, 2022-12-22:
I think having it configurable is actually a most, it can get annoying for some. I never tried it, it sounds cool in theory but in practice might be different. 

I was also thinking of having custom options per Di2 device, so you can beep before the synchro shift and things like that, but that is a bit too much to add now, so I will likely just make it a 'global' option for now. Beep on min gear/Beep on max gear.

Thanks for the input btw! 😄

robboyer, 2022-12-24:
An on/off control sounds good.  

If you do undertake this, it would be useful if the beep (when enabled) also happened when you tried to shift to a lower gear when you're already in the lowest, or to a higher gear when you're already in the highest.  Effectively it's telling you in this case "can't do that", which  is useful if you can't see the screen and haven't shifted for awhile.

valterc, 2022-12-24:
That is not possible Rob. We only get the current gear indication from Di2, there is no way (as far as I know) to get commands like shift up/shift down. So if you attempt to shift to a lower gear when already on the lowest there is no change in the Di2 data or any 'notification' indicating a shift down/up. 
The only way we know that there's been a gear shift is when the current gear also changes, this is how shift count works, for example.

akivanwasser, 2022-12-25:
Actually, valterc, I believe that the Edge beeps when you reach the highest gear or lowest gear, not when you try to shift afterward. Also, it has the option to beep in Synchro Shift and Semi-Synchro shift (S1 and S2) when you reach the gear before a chainring shift.

valterc, 2022-12-27:
I've been researching this a bit more and turns out that it's indeed possible to beep when you shift into the lowest/highest gear, when overshifting (shift down when in lowest gear/shift up when in highest gear) and also before the synchro shift (if enabled).

Will work on this for next update.

judahrand, 2022-12-27:
Worth deciding exactly when it should beep. 

For example, if I'm not using Synchro Shift Mode 1 and I shift into the biggest cog at the rear but I'm also in the biggest cog at the front should it beep? It is the end of the range I have unless I manually shift the front chainring. Open question. 

I can't remember what the behaviour is on Garmin?

bigfloor, 2022-12-27:
Behavior on Garmin is as follows:
- in Di2 you can setup the gearing for synchro shift where you 'lockout' certain gear combinations
- this is available for the front big ring, like locking out the 3 biggest sprockets on the back
- and also on the front small ring, like locking out the 3 smallest sprockets on the back
If you then reach the latest allowed combination, the Garmin will beep. If you then still press your Di2 button to switch gears, synchronized shift will use your Di2 settings and switch on the front (big to small or small to big)+ on the back (most likely 2 gears up or down) to achieve your desired gear increase/decrease change (which is tuneable via E-Tube)

robboyer, 2022-12-27:
I didn't use synchro shift when I had my Garmin, so can't comment on that.   I didn't change any of the default beep settings using E-Tube, and here's how I remember it working (don't have the Garmin anymore to verify):

It would only beep when going to the lowest or highest gear, or when in the highest gear and you tried to shift higher, or in the lowest gear and tried to shift lower.  It didn't beep when you went large-large, although that might be a nice enhancement since it's rarely desirable (maybe do 2 beeps for that case?)

Thanks for working on this!

valterc, 2022-12-27:
I'm not sure if we can beep in that scenario @judahrand, will need to test that out when riding.

I believe that is correct @robboyer. According to some documentation, the limit beep happens when:
 
> Gear shifting limit has been reached. (When the chain is on the highest gear for both front and rear or lowest gear for both front and rear).

For the synchro shift part, it happens when:

> The front gears are being shifted in shift mode and the front gear will be shifted in the next operation.

Question: Would it make sense to have individual setting for beeping when shifting into the overall lowest gear and another setting for when shifting into the overall highest gear? Or just combine the two and make it simpler "Beep when shifting limit reached"?

robboyer, 2022-12-28:
> Question: Would it make sense to have individual setting for beeping when shifting into the overall lowest gear and another setting for when shifting into the overall highest gear? Or just combine the two and make it simpler "Beep when shifting limit reached"?

IMHO a single setting would be fine and less confusing - and I like your wording.

valterc, 2022-12-28:
[image]

I ended up with that, to be honest still considering simplifying it to a single option for lowest and highest gear. 
Also still need to test it on the road, hopefully I get a sunny day soon.

closed 2022-12-29 by valterc

valterc, 2023-01-01:
This is now available in [release 0.9](https://github.com/valterc/ki2/releases/tag/0.9). The app should show a notification for auto-update or you can trigger an update via the settings.

Please give it a test when you go for a ride 🙂

robboyer, 2023-01-02:
@valterc tried it today on a 60km ride and it worked perfectly!  I used all the new beep features except "Upcoming Synchro Shift".  No issues whatsoever, and connectivity was solid.    I'll try on the Creo gravel bike soon (when the rain stops) to make sure it works when there's no front derailleur.

valterc, 2023-01-02:
Thanks for testing @robboyer!

judahrand, 2023-01-02:
I gave this a test this morning and I think I've identified a bug. When a ride is stopped and then a new ride is started the audio alerts no longer fire. A reboot fixes the issue. @valterc would you like me to open a separate Issue?

valterc, 2023-01-02:
Interesting, how did you stop the ride, just normal end ride and save/delete? Feel free to create a new issue

valterc, 2023-01-02:
Actually just found the problem, easy fix in #83. Thanks for the testing @judahrand! 🙂 
(Will likely group this fix in the next release, maybe by end of week, so until then keep this in mind)

PR #76 (2022-12-29): Add shifting audio alerts; Improved low battery notification
Resolves #70