valterc/ki2 #73: Feature request: e-bike support

SMXRanger, 2022-12-26:
In 2018, when I have Wahoo Bolt bike computer (and it does not support Japanese e-bikes officially), it was possible to pair e-bike with bike computer as Di2 sensor to view e-bike battery %. May be it is possible to add this functionality (with small amount of time investment) into ki2?

For now, Karoo 2 and Ki2 doesn't see e-bikes at all.

valterc, 2022-12-26:
Not sure if possible, but to be honest it's likely not worth it because Hammerhead is adding ebike integration soon: https://support.hammerhead.io/hc/en-us/community/posts/10948604813211

Closing as not really in scope for Ki2.

closed 2022-12-26 by valterc

SMXRanger, 2023-02-26:
You add support for STEPS in 2.0 version. Thank you! 

Showing battery status and current gear of Di2 shifter, connected to motor. 👍