valterc/ki2 #74: Two Simultaneously Connected Di2 Causes Looping between devices

haspden, 2022-12-26:
Opening a Di2 when two devices are connected Simultaneously causes the
- battery
- signal strength
- serial number
- model number
- gears
- etc

And other related data to flicker rapidly between the two active connections.

Clicking on the device appears to open all active connections not be filtered by decice ID / serial number of the one clicked upon.

See video attached

https://user-images.githubusercontent.com/4039041/209569989-848c2dd6-65e5-43c9-b721-e44097bca4f9.mp4

valterc, 2022-12-26:
Thanks for reporting 🙂 I was already aware of this problem and started working to fix it in #72. Still need to do some more work, until a new release comes, better to have a single device added/connected at a time.

haspden, 2022-12-26:
> Thanks for reporting 🙂 I was already aware of this problem and started working to fix it in #72. Still need to do some more work, until a new release comes, better to have a single device added/connected at a time.

Thanks @valterc , yes just noticed it today whilst adding a second bike whilst giving them all a nice winter strip down, clean and rebuild

valterc, 2022-12-28:
Fixed now, I still need to test other things on the road, it might take some days until it's released.

PR #77 (2022-12-28): Fix device details when multiple devices are connected
Fixes #74
closed 2022-12-28 by valterc