valterc/ki2 #75: Feature request: Synchro Shift notification

judahrand, 2022-12-26:
Garmin's implementation of Di2 support shows an on screen notification and beeps when upon the next shift Synchro Shift will shift the front derailleur either up or down. It would be amazing if Ki2 could do the same.

valterc, 2022-12-27:
Working on this for next update. As far as I can tell we don't need the synchro shift configuration, the shifting system will tell when the upcoming shift will happen on it's own.

judahrand, 2022-12-27:
That's fantastic to hear! Thank you!

judahrand, 2022-12-29:
Was this also closed by #76?

valterc, 2022-12-29:
I believe so, still need to go out to test it before releasing. Also only the is only beeping (audio), there is no on screen (visual) indicator.

judahrand, 2022-12-29:
> I believe so, still need to go out to test it before releasing. Also there is only beeping (audio), there is no on screen (visual) indicator for now.

Fantastic, very excited to try out the self-updater when this is released. The lack of visual indicator is probably the less useful indicator anyway (though I do find that the Karoo 2's beeper could be louder). Though, perhaps an option to add in the future. The preferences/settings might start to get complicated!

closed 2022-12-29 by judahrand

valterc, 2023-01-01:
This is now available in [release 0.9](https://github.com/valterc/ki2/releases/tag/0.9). The app should show a notification for auto-update or you can trigger an update via the settings.

Please give it a test when you go for a ride 🙂