valterc/ki2 #85: Bug: Hood buttons don't work until Ki2 data element opened

judahrand, 2023-01-03:
I had expected that even if I don't have a Ki2 data element on a ride profile the hood buttons should work if "Pre load ride application" setting is enabled. However, this does not seem to be the case - a Ki2 data element must be present.

valterc, 2023-01-03:
That is indeed correct @judahrand, for the hood buttons to work you will need at least one Ki2 data element in the ride profile. I can try to make them work anytime inside the ride app, but I cannot guarantee it will work reliably.

judahrand, 2023-01-03:
I see. It's not really a bit deal as one can just have a screen with just Di2 battery or something but it would be nice for it to 'just work'. I'm familiar with these sorts of limitations as they exist with some Garmin ConnectIQ applications.

PR #88 (2023-01-04): Support switches on ride profiles with no Ki2 elements
Resolves #85
closed 2023-01-04 by valterc