valterc/ki2 #86: Feature Request: Ability to rename Di2 sensors

robboyer, 2023-01-03:
If you have ride than one Di2-equipped bike and switch the Karoo between them, it can be difficult to tell which Di2 sensor corresponds to which bike, because the devices appear in the ki2 list as "Di2 <someRandomNumber>".  It would be awesome if you could rename the devices into something more user-friendly, for example:

"Aethos Di2"
"Creo Di2"

etc.  I believe the wahoo and/or Garmin computers let you do this, as well as the Karoo itself.  A logical place to add a "Rename" button would be on the screen that comes up when you touch a sensor in the list (e.g. where the "Remove" button is")

bigfloor, 2023-01-03:
Totally agree, I also have 2 bikes with Di2 (1x 8050 and 1x 8170), it's hard to remember which one is which.

valterc, 2023-01-15:
Will be available in the text version 🙂

PR #89 (2023-01-15): Device preferences and rename devices
- Add preferences associated to a specific device
- Add custom name for a device

Resolves #86
closed 2023-01-15 by valterc