valterc/ki2 #101: Data not showing in profile view with latest update from hammerhead.

rvico, 2023-01-24:
I updated karoo2 with latest software version 1.352.1409.12, and can't see any information on profile view. App is loaded and connects to di2 without ussues, but when riding or entering profile, the fields only display N/A or "..." (three dots), no information is visible from ki2 in ride.
I think hammerhead broke something, intentionally?

I have latest version of ki2 1.0. I already try to uninstall and install again, but same issue.

valterc, 2023-01-24:
I've been using the app on 1.352.1409 and haven't seen those problems, but I've seen something that might be causing that for some people, fixed in #99 but still not released.

If you didn't, restart Karoo after updating/re-installing Ki2. Do you by chance have multiple Di2 devices in the app?

rvico, 2023-01-24:
This is what I've tested:
Uninstall app.
Reboot device.
Install app.
Reboot device.
Tested... and again same problem. Now battery displays ok, but any gear indicator is working for me, it shows three dots only.

I've seen that when di2 connects, if I enter to see di2 details... I can see that there are two blocks "waiting for data", the top one takes some time but finally displays manufacturer information, but the second one at bottom is "waiting for data" eternally. Don't know if it is related with #99 . 

Thanks. Let me know if I can help in any way.

valterc, 2023-01-24:
I see, that is indeed different, in that case the problem is the ANT connection to Di2. 

It's normal that the manufacturer info takes a while to appear, the Di2 unit is not broadcasting that data often. The shifting data (which is the next block) on the other hand should appear almost immediately. I did a change in version 1.0 that the shifting data waits for chainring information, if we don't receive the chainring information then the data is not used.

Could you try Ki2 version 0.9 and check if that works? https://github.com/valterc/ki2/releases/tag/0.9

Also, do you have a Ki2 setup configured in E-Tube with the chainring information, like so?
[image]

rvico, 2023-01-24:
That seems to be the problem, my groupset doesn't connect with the japanesse brand app, its the very first one di2 groupset that brand put in the market... a bit old but works flawlessly. Can't configure the chainring information on it.

I try with v0.9 and it is working again. I will stick with this version for now.
Maybe you can set that information from the groupset as optional to increase compatibility with older groupsets, or add a parameter so that people with older groupsets can mask and set the rings manually (set gear 1 = 27,  gear 2 = 25...) like on garmin edge units.

Thanks in advance.

valterc, 2023-01-24:
Thanks for diagnosing and testing! 

Ok, please continue to use 0.9 for now, I will make a patch to fix this issue for the next update. I like your suggestion to be able to configure the chainrings manually, it's something that I considered but that will take some time, something for a later app version.

robboyer, 2023-01-24:
> Hi Rvico
> I had similar issues when trying to use version 1.0.
> I rectified the problem by factory resetting my Karoo 2 and then reinstalling version 1.0.
> It now runs flawlessly.
> Hope you get it working.

FWIW I’ve been using v1.0 with 2 di2 setups and have had no issues.  I didn’t have to reset the Karoo...

I think both my di2 setups work with e-Tube, so maybe that’s why I’m not seeing any problems.

valterc, 2023-01-24:
@e17biker What issues did you had? Di2 connects fine in the app but no data is available in the ride elements?
It's weird that it works fine after factory reset... you really shouldn't need that, unless there was some configuration causing issues. Do you have the exact same setup after factory reset?

valterc, 2023-01-24:
Issue from here is now patched in #103, will be available in next app update.

e17biker, 2023-01-24:
Hi Valterc

First let me say a big thanks for taking time to find a solution to a problem that's shouldn't exist, ki2 is fantastic!!!

Yes I had connection but no data in ride elements.

Tried uninstall/reinstalling, deleting sideloaded app all to no avail hence the factory reset.

Since then I have had no problems, same exact setup.

Thanks again for you time making this app. 🙏🏾

valterc, 2023-01-25:
Thanks @e17biker!

I find it weird that you had to do that, but I'm glad you made it work!

If you have further problems please open an issue and we can try to diagnose and eventually fix the issue, I think it's not that fun to keep doing factory resets if that happens again.

valterc, 2023-02-21:
The fix is finally available in the newest version [2.0](https://github.com/valterc/ki2/releases/tag/2.0). Please install/update.

closed 2023-02-21 by valterc