valterc/ki2 #106: Bug: ki2 gear widget on 1 of 2 data pages within profile doesn't update

robboyer, 2023-01-26:
I created some new data pages within my bike profile recently, and 2 of the pages have the "Gears" ki2 widget.

Today, on a ride (first time using this profile), the ki2 "gears" widget on only one of the data pages updated as I shifted gears, the other one did not and was stuck on 1/2 (front) and 6/11 (rear) the entire ride.  Note that the graphic elements were showing, but they simply didn't update.   Shifting while on the page did not cause the widget to update.  The other data page with the same ki2 widget updates file, whether or not you're on that page when shifting.

Strangely, while on the non-working data page, the audible alerts at shift limits still worked fine.

valterc, 2023-01-26:
You are absolutely correct, in fact I'm surprised it was even displaying both elements. 
Need some QA testers for these edge cases 🙂

closed 2023-01-27 by valterc

valterc, 2023-01-27:
I've fixed this but still need to do more testing to make sure it does not consume too much battery and works on long rides. 

Karoo SDK is really not optimized, it tries to create all these widgets many times over and over during a ride, so before this change I was giving Karoo always the same widget, but that does not work if you actually need two (or more) of those widgets, for example when you have the same widget in multiple pages. 

If the tests are OK the fix will be available in next update, otherwise we might have the limitation of max 1 of each graphical widget per profile.

PR #110 (2023-01-27): Fix same graphical view in multiple data pages at once + service client refactor
Fixes #106