valterc/ki2 #107: Unable to add my Di2 system.

HenryBlack1, 2023-01-26:
Firstly thanks for all your hard work. I've got a Giant ebike equipped with Di2 shifters. Single chainring, 11 cogs on the back. The right hand side shifters take care of the rear gear change whilst the left hand side shifters look after the power assist levels. The buttons on top of the hoods do nothing, but I'd love for them to control screens on the Hammerhead2. 

Details.
Giant Revolt E+
Sensor-RideControl GO, frame integrated, Support up/down from STI levers
Shifters-Shimano GRX Di2 ST-RX815-R/L 1x11speed

So I've somehow managed to bumble my way through side loading the app onto the Hammerhead2 unit. I thought that would be that hardest part done. I can open the app. Searching for the device brings up Sensor 64475, but there's no "ADD" button off to the right to push. 

[image]

It seems to me (a technical dummy) that the Di2 sensor can be seen, but I'm not able to connect to it. 
The Karoo is only new, it's still in the 45 day satisfaction period, so if I can't get this sorted I can send it back and grab a Garmin/Wahoo. I'd much rather keep it though. 
Any help would be great. Cheers.

valterc, 2023-01-26:
That is happening because that unit is broadcasting in the right frequency but it is reporting a different device type. Are you using a Di2 wireless unit like the `EW-WU111` or are you connecting to some other sensor? 
It might be a different Di2 spec that I'm not aware of or it might be a different profile that is not compatible with Ki2.

HenryBlack1, 2023-01-26:
Valter, Thanks for your quick reply.
I'm using the Shimano EW-EN100 wireless junction. Is there a way around it,
or is the Karoo incompatible?

Thanks.

valterc, 2023-01-26:
Did some quick research looks like the `EW-EN100` is not a 'pure' Di2 wireless unit, but rather a "shimano steps", I'm not really sure if it is fully compatible with Di2 functions. 

I can try to make a custom Ki2 version with some changes to allow you to try to connect, but I don't know if the app would be able to communicate with the unit properly. I don't have any e-bike with that unit to test with.

HenryBlack1, 2023-01-27:
Valter,
1. Do you think the current ki2 would work if I replaced the EW-EN100 with the EW-WU111, or is there any other Shimano unit you feel would work?
2. I'm guessing this means the current setup wouldn't work with a Garmin or Wahoo computer either, as far as the functionality of the hood buttons?
3. Would there be much work in a custom version, or would it be just some simple changes? Is it something I could test from here?
Thanks heaps for your help And your time.

valterc, 2023-01-28:
1. Yes, Ki2 works with `EW-WU111`, **but** I'm not sure `EW-WU111` is fully compatible with e-bike systems, so it might or not work with your Di2 e-bike setup, research or test before committing to do that.
2. It's likely that Garmin/Wahoo have support for Shimano Steps/`EW-EN100`, check with them. They have access to the protocol specs, so don't see why they won't support it.
3. I've made this custom version: 
[ki2-devtype_exp-release.zip](https://github.com/valterc/ki2/files/10528031/ki2-devtype_exp-release.zip)
This version should allow to try to connect (the ADD button will be there), but I'm not sure what will happen after that, it might work just as fine or not at all. 

- Download that zip file
- Extract/Unzip that file
- You will have the APK file inside the extracted directory
- Connect Karoo, open terminal
- Uninstall the existing Ki2 version (you can do this from Karoo, hold the Ki2 app and press delete or uninstall via the ADB command, similar to the install process)
- Install the APK that you've extracted previously from the downloaded zip file

Let me know if you try it and please send the Di2 name that appears in the list, should be the same name 64475 with some characters added after.

HenryBlack1, 2023-01-29:
Ok. Firstly thanks again for your time with this.
Uninstalled the old, installed the new version.
App appears in the right place. Added.
[image]
Searched for Di2. Connecting
[image]
Di2 64475-2 Not Found.
[image]

I've tried to put the EW-EN100 into a "Connection Mode", if there is one, by pushing or pushing and holding the button on it. But this only seems to put the rear derailleur into recovery mode.
One step closer I guess?

Thanks.

valterc, 2023-01-29:
Thanks for the detailed info. One more version for you to try:
[ki2-devtype_exp_2-release.zip](https://github.com/valterc/ki2/files/10530344/ki2-devtype_exp_2-release.zip)

Hope it connects now, otherwise not so sure if possible. Give it a try, you will need to uninstall and install again, and let me know how it goes.

valterc, 2023-01-30:
Great! 

> Had to flick some levers to wake up the system?

Yes, this is normal. Same for other Di2 systems, need to wake them up.

I cannot see the photos, probably you haven't shared/allowed public access? I don't need to see them anyway, I was looking for the numbers in the device name, so it's `64475-2-5`. 

If you open the device details, by pressing on the device name in the device list, which information do you get there? Battery + manufacturer info + gearing + switchers/button status? Is the information there correct, like the current gear? Are the buttons working fine?

To be honest I haven't changed much about the logic, the only thing is that before the app was expecting always Di2 with ANT device type = 1, this is how it detected if it was a valid Di2 system or not. Seems like this Shimano Steps/`EW-EN100` is device type = 2, so I just had to make it dynamic to support that, mostly everything else is the same. 

If you confirm all is working fine:
 - Battery status - not sure if you will get the e-bike battery or the Di2 battery status - or is it just a single battery?
 - Manufacturer
 - Gearing
 - Switches/Buttons

I can change the app so that you'll be able to install the 'normal' Ki2 release after the next update.

HenryBlack1, 2023-01-31:
Pictures should be good now, my error, yep the device name is 64475-2-5.

[image]

[image]

valterc, 2023-02-05:
Added proper support for `EW-EN100` in #115. You will be able to install the normal Ki2 app from the next update (not available yet).

HenryBlack1, 2023-02-05:
Good man, well done & thanks again.