valterc/ki2 #120: Number of Syncro Attention Beeps

dcquad, 2023-02-14:
The default number of beeps seems to be 3. could this be reduced?  I appreciate having an audio reminder of the upcoming shift etc. but the default number is tooo many! I also use the varia radar and the combo is overwhelming. I'd like just one beep for the di2 integration.

thnx, Doug

valterc, 2023-02-14:
Unfortunately this is not really possible. 
I could perhaps add an option to choose the beep pattern, but the choices are limited to what Karoo provides. So we cannot easily just beep once, we need to 'play' an existing beep pattern.

ttalonso, 2023-02-18:
For me the current beep is perfect. Please, do not touch!
In any case, always giving the user the option is welcome!

PR #134 (2023-03-11): Improve audio alerts
Closes #120
closed 2023-03-11 by valterc