valterc/ki2 #124: Garmin remote like Titan Lab Kommander...

ttalonso, 2023-02-23:
I have seen that in ki2 v2.0 the option to use a remote control to pass screens has been added. This titan lab model is out of stock and I have a Garmin remote control. Is it not possible to add it as a sensor? It would be amazing!!!

valterc, 2023-02-23:
I had the same request just a couple of days ago on #122. In short I won't work on this for now.

- You should request that to Hammerhead, post and ask on their community forum: https://support.hammerhead.io/hc/en-us/community/topics
- Kommander is a special case because it simulates being a Di2 system, so it's like handling any other Di2 system.
- I don't have a Garmin Remote.

closed 2023-02-23 by valterc

ttalonso, 2023-02-23:
We will wait. Let's see if hammerhead ends up giving support. Thanks for your answer