valterc/ki2 #126: Use a compiled version

lockevod, 2023-02-28:
Hello! I know there is one missing file. I have the key for the ant, but I use in other languages. Generally I use something like this 

const unsigned char ANT::key[8] = { 0xTT, 0xTT, 0xTT, 0xTT, 0xTT, 0xTT, 0xTT, 0xTT};

Obviously it's a false and incorrect key... but i don't know what is the correct format in your missing file (I've generated the file, the problem is the "format" of the key... if I put like the example I have an error (because it's not Base64.. but if I use this key (0xTT,0xTT, etc... codified in Base64, the app fails because the key hasn't the correct format ... The key is correct and provided by thisisant ,etc.

Could you help me?

Sorry for ask for this hear but I don't know howto solve.

lockevod, 2023-02-28:
Sorry... I see i need the 128 bits key.

valterc, 2023-03-01:
Hi, sorry I cannot discuss that here and I will delete this issue. The key is 128 bit encoded in Base64.

closed 2023-03-01 by valterc