valterc/ki2 #132: Question about hood

lockevod, 2023-03-05:
Hello valter, your app is so great I’ve bought you a coffee ;) and thanks for your support in last two questions ;) 

I’m trying to provide edge remote support (I know isn’t in your roadmap); I think I can achieve with your work in ji2, but if you guide me how can map de buttons/hood with the buttons in Karoo this wil be very helpful 

I know the Karoo SDK is very limited and it’s not possible to send key commands and this will be possible in ride app, but if you can provide me where is done or some kind of guide howto do in ride app, it’ll be enough and very useful for me.

I give you my email (lockevod@pm.me) if you want to write there and no dirt these comments. ;) thanks a lot

lockevod, 2023-03-05:
Or if you want I can write you to your donation email ;) regards

valterc, 2023-03-05:
Thanks! :) 

The buttons in Karoo are actually simulating Keyboard Key Events, I have a mapping in file https://github.com/valterc/ki2/blob/main/app/src/main/java/com/valterc/ki2/data/input/KarooKey.java

```java
    /**
     * Left top key.
     */
    LEFT(KeyEvent.KEYCODE_NAVIGATE_PREVIOUS),

    /**
     * Right top key.
     */
    RIGHT(KeyEvent.KEYCODE_NAVIGATE_NEXT),

    /**
     * Right bottom key.
     */
    CONFIRM(KeyEvent.KEYCODE_NAVIGATE_IN),

    /**
     * Left bottom key.
     */
    BACK(KeyEvent.KEYCODE_BACK),
```

So you need to send these android physical keys to the Ride app. You can use some input systems like the `InputManager` ([this is what I do in Ki2](https://github.com/valterc/ki2/blob/main/app/src/main/java/com/valterc/ki2/input/InputAdapter.java)) from Android or grab the activity and use `dispatchKeyEvent`, etc.

These keys can also do several actions, depends if it's single press, double press or hold.

lockevod, 2023-03-05:
Thanks for your support. I’ll try to follow your examples ;)

Regards

closed 2023-03-05 by valterc