valterc/ki2 #137: No data displayed for gear count, Gears, shift battery,

Sirkaos56, 2023-03-17:
Version 2.0, I do have drivetrain and it works well and really like that display. Hood buttons are working well and I checked that buttons only was not selected. 
Side note, are you in Ukraine? I spent quite a bit of time there about 10 years ago, various cities. Loved it.

valterc, 2023-03-17:
Hi, if you open the details of the shifter in Ki2 do you see battery % and gearing information there? Should be similar to this image:
[image]

Keep in mind that in Karoo Ride Profiles, you need to replace all data elements with the ones from Ki2, for example if you have `Shf Bat %` from Karoo it won't show any data from Ki2, you need to use the element from Ki2 in the profile - even if the name is the same.

Post a picture of the details in Ki2 and the ride with the data problems.

Sirkaos56, 2023-03-19:
As you described, I have replaced all the data elements from Ki2. The shift battery % shows up when I open up the connected Ki2 menu but does not display in the data fields. I currently don't have Shift battery %, shift count, Gears or similar. I do have drive train and hood switch functions which is fantastic. Any ideas?

valterc, 2023-03-19:
No ideas. Those fields use the same information that the Drivetrain uses, so that's really weird. Can you post a screenshot/picture of the ride elements and the Ki2 gearing information?
What is displayed in the ride elements in 'Shift battery %', 'shift count', 'Gears'?

Sirkaos56, 2023-03-20:
I downloaded your 3.0 update and it looks like everything is perfect! All the missing data fields suddenly populated.  I didnt change anything other than doing your latest update. No idea why.

valterc, 2023-03-26:
Ok, thanks for updating :)

closed 2023-03-26 by valterc