valterc/ki2 #141: Add compact overlay theme

valterc, 2023-03-19:
> Small request: the overlay takes a lot of room from the map. It's it possible to allow used to choose which info to show? for example, just the gear image and then remove everything below that?

ttalonso, 2023-03-19:
I usually use the default map page. For me it would be very useful to have a few seconds the gear info. But since I have a single chainring on my bikes, only the rear Gear with a single number, type: 12, and that it was not intrusive with the map navigation would be very useful.

[image]

valterc, 2023-04-10:
Added a few more overlay themes and opacity setting, so now you could have something like this:
[image]

PR #153 (2023-04-10): Add compact overlay
Closes #141 

[image]
closed 2023-04-10 by valterc