valterc/ki2 #156: Di2 Connection lost during rides but doesn't come back

DMG080, 2023-04-11:
I'm relatively new to this and started with the 3.7 version a few weeks ago (just got the Karoo v2) - updated to 4.0 on 4/10.

The Di2 info is available during the first part of the ride but then disappears (shows N/A on the data and graphical drivetrain fields).  I believe this also happened on 3.7.

Sensors sometimes go in/out of connected status like the Garmin Varia but generally come back.  I wasn't sure if this was reported so thought I would check.  Thanks.

valterc, 2023-04-12:
I've had reports about that but haven't been able to replicate the issues.

- Can you tell me which Di2 version do you have (R7100, R8050, etc.) and if you have a wireless unit like EW-WU111?
- Do you notice the connection drops happen when you do something like not change gears for a while (several minutes), during a pause or any other action? 
- Does the connection to other sensors also drop/reconnect at similar time?

If that happens again, try to open Karoo control center and press the "Reload" button on the sensors tab and see if it reconnects.

DMG080, 2023-04-12:
Apologies for the delayed response.

- I have the 8100 series Ultegra Di2 (2022 Cervelo R5) => 8170 shift/break levers, 8150 rear derailleur
- I haven't noticed it dropping due to inactivity since I shift often enough once my rides start.  I mainly use either the Map screen (have gear numbers field e.g. "02-06") or lately more of a graphics layout with the drivetrain & upcoming elevation (two data fields are on top).  I've switched between these screens and a few others (8 total in the profile I use).  
- Yesterday (4/11), the Di2 info showed up and was present for a several miles.  Around mile 9 or 10, the data dropped again but this time it did come back - may have been missing for a few miles but wasn't most of the 32 mile ride.
- The Garmin Varia radar definitely drops several times during rides but always comes back.  This also happened when I had a Wahoo Roam v1 so it's not a Hammerhead Karoo 2 problem alone.
- I have some rides tomorrow and Saturday for sure so I can update results as I complete them.

Regarding the reload option for the sensors, I don't think I have the Di2 added there specifically (will check).  I have it specified in the Ki2 app since that feeds the info to the screens.  Maybe that's part of the problem?

nik-taylor, 2023-04-15:
I'm seeing this issue on a new R8100 Ultegra di2.  On my old bike with R8070, it was working fine when the DFly was sat pretty much under the Karoo.  On the new bike, i'm seeing a dropout every 20-30 mins.  It eventually comes back but it could take 1-5 mins.

kintrupf, 2023-04-17:
This happened to me yesterday during a ride. The Varia radar and the Di2 dropped the connection at the same time, so I guess it was something with the ANT+ in the hammerhead unit. The Varia reconnected automatically a few seconds later, but the Di2 did not. However, I was able to reconnect the Di2 manually a few minutes later in the ki2 settings.
I have an older Di2 with BT-DN110 battery, EW-WU111 wireless unit and DT-R8070 shifters.

DMG080, 2023-04-17:
Quick update on my most recent rides (4/13 & 4/15).  I continue to mainly use the Map & Graphical screens as noted in my original post but during the 4/15 ride, I switched to several other screens.  Neither ride showed a drop in the Di2 connection so that was good.

The Varia continues to drop and quickly reconnect so that's nothing new.  

I'll be busy most of the week but plan to ride towards the weekend.

valterc, 2023-04-17:
Tricky to understand where is the problem, I did find some occasions where Karoo resets the ANT service. That is my only guess so far, I was not yet really able to replicate the issue.

@kintrupf do you by chance remember how the disconnect happened, did the Ki2 data get 'stuck' or - if you had the overlay active - did it show "Connecting" or just "Disconnected" immediately?

nik-taylor, 2023-04-17:
Do you have a debug mode I can enable. I'm savvy enough to pull files down
and forward if that helps?

robboyer, 2023-04-29:
> Nik, there's no great way to debug this. One way would be for Ki2 to record everything while in ride, all ANT transmissions and any state changes, but it does not support anything like that.
>
> I've made some recent changes, as far I know there is no problem with the connection to Di2, the potential problem is in Karoo software that sometimes resets the ANT service/module and some other quirks with Android services.
>
> Might also be that some Karoo versions have different ANT modules? Just a guess at this point.
>
> Will do some rides with the changes to test.

FWIW I’ve tested on 2 different bikes on long-ish rides and had no dropped Di2 connections - everything is working great.  

This is a great piece of work and I really appreciate having it!  I especially like the pop-up gear display after shifting - very useful!

DMG080, 2023-05-06:
Since I reported the issue, I’m not losing the Di2 connection anymore
(knock on wood).

I just upgraded my Karoo 2 with the latest firmware update today before my
ride (1.383.1499.12) and things still work well.

valterc, 2023-05-06:
Thanks for the feedback. 

I've made some more changes to improve connection problems, still need to do some more rides to make sure all is good. 
Will release the update in a couple of weeks with some other changes.

closed 2023-05-06 by valterc

Allanhaapa, 2023-06-06:
Hi i have issues with connection. It drops after 3-4 minutes and not reconnect. I have your latest update. I use ew-wu111.

nik-taylor, 2023-06-06:
I've had much better success with 5. Thanks very much for the fixes. It
still drops a few times an hour but comes back much quicker.

valterc, 2023-06-07:
The is very little else I can do in the code, I was never really able to reproduce such intermittent issues, I suggest folks with problems to try to troubleshoot:

- Make sure you have less than 14 ANT sensors configured in Karoo
- Make sure all sensors are disabled/not connected except for the Di2
- Open Ki2, tap on your Di2 device:
  [image]
- On the device details check the signal value on the top right:
  [image]
- Setup Karoo on the bike and go for a quick ride, how is that value changing? Is it around -40, -50 or -70, -80? Signal less than -70 are degraded and connection could drop easily - especially with interference from environment/other sensors.

If signal is all good and within reasonable values (-40 to -60) then try starting enabling other sensors and do the same checks. 

Typically, if you can, I would suggest moving the Di2 unit (ew-wu111) closer to Karoo.

Allanhaapa, 2023-06-07:
Hi again. I have tested it properly now. I have missed to change to Bluetooth in karoo2 settings  sensor. It working perfect now. No communication problems under 100km ride.

PR #166 (2023-05-06): Improve connection
- Increase connection retry count
- Add ANT Service auto re-bind when service connection is closed
- Restart connections in state 'connecting' so that connection timeouts gets reset
- Decrease time in between service re-binds

Closes #156