valterc/ki2 #172: Add Shimano EP8 Support Please!

squier79, 2023-05-13:
hi, it would be awesome if you can add support for shimano ep8 motors. Ebike already connect to your awesome plugin, but there aren't fields to add to karoo like Battery Level, assistance mode and remaining km. With these 3 fields there will be an awesome feature in this plugin.
Thanks for your attention, have a good day.

valterc, 2023-05-13:
I cannot really add support for EP8 (STEPS) specific fields (assistance mode, remaining km) because I don't have access to a bike with EP8. I would need personal access to bike with EP8, if anyone has such a bike in Europe, we could try to meet.

Battery level should be available, it is only in text with the name "SHF BAT %" - this will show the current EP8 battery %. For the next Ki2 version (not yet available) I've also added a new graphical battery view similar to the new Karoo e-bike views.

Version 4.0:
[image]

Next version:
[image]

squier79, 2023-05-13:
I’ve tried to connect bike (connects really well and fast) and see the battery level but remains empty… I need to do something more? I have an ORBEA rise H15 2022 with shimano ep8 RS

valterc, 2023-05-13:
What do you mean by "remains empty"? Does it have any text? Where are you checking the battery? In the Ride app?

To see the EP8/STEPS/Di2 battery you need to use Ki2 elements in the Ride profiles, edit a profile and replace the Karoo text/graphical elements with some from Ki2, for battery you need to pick this one:
[image]

You should also be able to see the battery level directly in Ki2. From the main Ki2 screen, press on your Di2/STEPS device and in the new screen scroll down until you see the battery and gearing information. Does is show there?

Please post a picture of the problem if it still doesn't show correctly :)

squier79, 2023-05-13:
Hi! I don’t know how but I did it! I have battery now! (maybe because I’ve restarted my Karoo) Thanks so much! Hope for the new version with the graphical battery view! 😍
These are 2 screenshots of my Karoo 2 connected with my ORBEA rise

SMXRanger, 2023-05-15:
I'm using Ki2 with EP8 too, and it works well, thank you! I have one request: is it possible to add a new option for Ki2 to search disconnected sensors all the time?

The problem is: when I pause my ride to buy a coffee, motor goes to sleep and Ki2 loosing connection. When I get on bike again and power it up, I need to open Ki2 and tap "reconnect" button.

squier79, 2023-05-15:
@valterc Goodmorning, do you know when will be available the new version with the graphical field for the battery? Thanks!

valterc, 2023-05-15:
@SMXRanger I did some changes recently that should help that. Ki2 will now attempt to restart connections after resuming a ride (although it likely won't work on auto-pause/auto-resume because of Karoo bugs)

@squier79 Probably next week, still need to test some other things first.

valterc, 2023-05-21:
New battery fields are now available in Ki2 version 5.0.

closed 2023-05-21 by valterc

squier79, 2023-05-21:
Wow thanks! 😍😍😍😍

squier79, 2023-05-21:
Just updated… but I can’t see % of battery anymore. Ep8 shimano

valterc, 2023-05-21:
Restart Karoo after updating. 
To use the new fields you need to customize a Karoo profile with elements from Ki2.

squier79, 2023-05-21:
I’ve restarted the Karoo2… and added the field. I sent a screenshot did you see it?

valterc, 2023-05-21:
No, sorry, you need to post the screenshots in Github directly, replying to email does not send the screenshot.

squier79, 2023-05-21:
[image]

ok sorry here is the screenshot

valterc, 2023-05-21:
That is maybe a difference of the EP8 system, perhaps it only broadcasts some data when active/riding? I'm not sure how that works, from that screenshot I can see it is connected but hasn't received the battery value yet.

squier79, 2023-05-21:
With the 4.0 version it works…

valterc, 2023-05-21:
Ok, you need to help me understand what is the problem, could you re-install/update version 5.0 and:

- After install, open the ride again with those elements and plug you Karoo into a computer and can you run `adb logcat` in your computer terminal. Just like you did `adb install` but this time do `adb logcat`. Please paste here the output.
- Do you have the latest Karoo software? It should be version 1.386.1509. You can check in Karoo Settings, About.

squier79, 2023-05-21:
Ok tomorrow I’ll send you all. Thanks for help

squier79, 2023-05-21:
Ok it works! Just removed e-bike and added again, then restarted Karoo and after restarted e-bike to. Now it works! Thanks a lot!

valterc, 2023-05-21:
Ok, if that happens again let me know and we can try to understand what is causing the issue.

squier79, 2023-05-21:
Ok thank you so much