valterc/ki2 #185: Cannot build app from Source

NordicFlyer1, 2023-06-25:
There seems to be something missing in the source code (a file or network key). Please advise. Cheers!

NordicFlyer1, 2023-06-25:
Never mind...I figured it out. Thanks.

NordicFlyer1, 2023-07-02:
Source code will not build. Does this have something to do with the private ANT key? Error message:
 error: cannot find symbol
        byte[] networkKey = Base64.getDecoder().decode(context.getString(R.string.network_key));
                                                                                 ^
  symbol:   variable network_key
  location: class string

valterc, 2023-07-03:
Hi, yes, this is expected and documented in the FAQ:

> I tried cloning the repo but the build fails on my machine
> 
> Yes, this is by design. There is a particular file missing from this repo and unfortunately I cannot add it here.

I cannot add the file here, if you want to build the app you need to find the file/value from some other source.

closed 2023-07-03 by valterc