valterc/ki2 #186: Karoo not recognizing Ki2 in data field

Raseyeguy, 2023-07-12:
Ki2 was working well. Now it is not recognized in data field. Ki2 is connected but no info shows up in data field. 
[image]
[image]

Thank you.

valterc, 2023-07-14:
Hi, I haven't experienced any problems like that, a few question to help troubleshoot:

- When did that happen? Middle of ride, when you where starting a ride, after a pause, etc?
- Did it eventually start working again?
- Have you tried to restart the Karoo?
- Do you have the latest software version 1.395.1557? Or do you have a testing version?
- In Ki2 open the device that is connected, scroll down, does it show the battery, manufacturer and gear information there?

Raseyeguy, 2023-07-14:
It happenedSent from my iPhone

Raseyeguy, 2023-07-14:
It stopped working after I turned Karoo off for an hour, and then turned back on and selected resume ride. Ride did NOT resume, and I lost the ride that I did prior to turning off Karoo. Unit otherwise functioned normally, but I was unable to get it to recognize Ki2/Di2. Ki2 did see the Di2, recognizing the gears and battery. Despite turning Karoo off and restarting multiple times, I am still not able to get Ki2 data in the data field, even though I have selected ki2, then gears for that data field. Both Ki2 and Karoo have latest updated software. Thank you. Robert SklarSent from my iPhone

Raseyeguy, 2023-07-16:
Karoo was not used for a few days. Rode today, and Ki2 still not being recognized in data field. I have selected Ki2, gearing for the graphical field, but it just says “waiting for data”. When I look at Ki2 app, my di2 is connected and shows correct gear position and battery. I believe I have the latest software version, but it’s different than what you describe-Can’t figure this out….it started after I turned Karoo back on after not ending a ride. Karoo is working correctly otherwise. Thank you. I have really enjoyed using Ki2 and hope it will start working again. Robert Sklar

Raseyeguy, 2023-07-23:
I uninstalled the app, and then installed it again. Works perfect, once again!Robert Sent from my iPhone