valterc/ki2 #187: Connection loss when on mount (shimano r8000)

TanguyCme, 2023-07-13:
Hi @valterc ✌️
First of all thanks a lot for this tool, really easy to use and lot of useful features ;)

I’m facing an issue with my r8000 groupset,
When I connect ki2 with it putting the Karoo just near the rear groupset, it works perfectly fine, something around -57db. But, when I put the karoo on the bar mount, it’s a mess, never even a signal, I assume that is due to a pretty poor ANT signal from the rear of the bike…

So my questions are :
Is there a way to boost a bit the Shimano Ant signal ? 
is there a way use Bluetooth that should be powerful ? If not, is there a way to be able to collaborate ? I would be really happy to help :)

valterc, 2023-07-14:
Hi, thanks!

What is the signal power around the bar mount? Typically `-57db` near the groupset is already not that great, do you a lot of other equipment which may be causing interference?

We cannot do anything to improve the signal quality from software. You could try to implement Bluetooth communication but it's a very different protocol, you would need to totally re-implement the communication with Di2 because data is handled in a very different way - I cannot really help on this part.

My suggestion is to try to check the settings in e-tube, maybe disable Bluetooth communication and use only ANT, and check for possible interference causes. Maybe it works better outside the house in open road?

TanguyCme, 2023-07-14:
On the bar mount it keeps trying to connect 90% of the time, when I start the Karoo next to the rear of the bike I can reach at best the value below
[image]

I know that it pretty low but here is the best signal I can have when the bike computer is just next to the rear of the bike
[image]

And no other equipment instead of my phone connected via BT :(

valterc, 2023-08-05:
Sorry but I don't think we can do anything in Ki2 to improve that, it's a signal issue 🙁 

You could try to disable Bluetooth on Di2 with E-Tube and/or try disabling the Bluetooth on the Karoo (disconnect you phone from Karoo) and see if it improves, otherwise I don't have any ideas.

closed 2023-08-05 by valterc