valterc/ki2 #188: triggering switch does not wake screen up when in battery save mode..

proseLA, 2023-07-30:
first off, THANK YOU!

second, in the firmware that i ran from karoo when the integration was still there, i could wake up the karoo screen by triggering one of the switches on the top of my shifting hoods.

i frequently run my karoo in battery save mode so this was an important/nice/necessary (chose one) feature.

now, if i am in battery save mode and i trigger an event from the hood, the screen does not wake up.

i have confirmed that when the screen is on, the events from D-Fly channel 1 and 2 both work as expected.

it would be great if the trigger event would start by waking the screen up no matter the event.

i am running 1.395.1557.12; it is probably not (yet) supported by you; but i was forced into a new device due to crash with the old one, and was hence forced to update.

i would attempt a PR; but my java skills are basically non-existent.

thanks in advance.

jeremywj, 2023-07-31:
I'll second this request.

valterc, 2023-08-05:
Hi, thank you!

That is very nice suggestion, I will work on that when I have some spare time 🙂

valterc, 2023-09-05:
Still need to do some testing. This should be available in the next Ki2 update.

closed 2023-09-05 by valterc

proseLA, 2023-11-21:
any idea when the next update might be?  curious minds are....  curious!

valterc, 2023-11-22:
Hi, sorry for the delay. This part is actually done but I want to finish some other things before releasing the update. Unfortunately I didn't have much spare time on the last few months for that, but thanks for the nudge. I will try to release it soon :)

PR #194 (2023-09-05): Turn screen on via switches
Closes #188