valterc/ki2 #189: Connection loss

JPAC-PT, 2023-08-02:
Hello. 
In this last two version, my di2 are losing connection and reconnect again

jeremywj, 2023-08-03:
JPAC-PT, are other sensors dropping as well?  Starting with the Jun 29th update I've had a handful of cases where all sensors dropped (including Ki2).  I have not had any instances of just Ki2 dropping.

JPAC-PT, 2023-08-04:
No, only ki2 sensors. The other sensors never lose connection and work properly.

valterc, 2023-08-05:
Hi, you need to debug it yourself and give more information, most likely it's a signal issue between the Karoo and the Di2 wireless unit, please follow the guide on the readme:

> The app disconnects from the shifting unit while riding
>
> Make sure you are using the [latest Ki2 version](https://github.com/valterc/ki2/releases/latest/). It could be a signal issue, verify the signal between the wireless unit and Karoo. To do that, tap on the shifting device on the list of devices in Ki2. The signal value is displayed on the top when the shifting unit is connected. Signal values between -30 to -60dBm offer best and stable connectivity while a signal in range of -75 to -90dBm is degraded and can cause connectivity issues. There is nothing that can be done in the app/software to improve the signal. Recommendation is to move or rotate the shifting wireless unit, perhaps moving it a small amount could be enough or in some cases it may require relocating the wireless unit to another location in the bicycle closer to Karoo.

If the signal is good, it might be some recent change to your setup.

- When you say "last two version, my di2 are losing connection" do you mean Karoo software version or Ki2 version?
- Did you add or change any sensor to the bike (watch, hr, power, etc...) or connected your phone or headphones to Karoo via Bluetooth?

closed 2023-08-28 by valterc