valterc/ki2 #190: Gears in graphical format, not in overlay but in data field

Etscod, 2023-08-08:
HI Valter, I just installed Ki2, excellent work and paid coffee! I apologize for my very bad English.
I would like to be able to display the gear graph in a double data field (for example, like the speed field in the first photo), whereas now I can only see the gear (or teeth) number. I've tried using an overlay with infinite time (second photo), but it takes up more space than an underlying field and obscures other information...
Do you think it's feasible? is there any other way to do it? Thank you very much!
Stefano

[image]

[image]

valterc, 2023-08-08:
Hi, thanks for the coffee! 🙂 

To add graphical elements you need to configure that Karoo profile and choose a layout with a graphic element. More information: https://support.hammerhead.io/hc/en-us/articles/1260805496570-Karoo-2-Graphical-Data-Fields

Edit a profile:
[image]

Edit layout, you need to choose a layout with a graphical element, any one with those charts lines will work.:
[image]

Choose Ki2 graphical element:
[image]

[image]

Etscod, 2026-02-16:
Hi Walter, thanks for the quick reply! I solved the problem by uninstalling the Ki2 v.8.1 app and installing v.14. It seems to work fine, thanks!