valterc/ki2 #191: Ki2 on K1?

realsimon, 2023-08-18:
Hi Valterc, I killed my K2 and while waiting for a new unit, I found an old K1. I understand that you wouldn't want to waste time supporting K1, but is it possible to load it as an experiment? I get "INSTALL_FAILED_OLDER_SDK" trying to sideload. AFAIU I would need a build without limit in android:targetSdkVersion for this. Or it's not going to work anyway?
This is more out of curiosity then necessity, so no worries if you don't want to deal with this.

valterc, 2023-08-19:
Hi, there is a special version for K1, you can find it here: https://github.com/valterc/ki2/releases/tag/0.8-karoo1

Keep in mind that Ki2 version is quite outdated and no updates are planned as the K1 is out of support. It does have all the basic features, so it should enough for a while.

closed 2023-08-19 by valterc

realsimon, 2023-08-21:
That's great, thank you!