valterc/ki2 #192: Suggestion: add screenshot function for support purposes

kintrupf, 2023-08-21:
Just a silly suggestion, but it would make it quite easy to create a clear screenshot for support purposes.
I assume here that it is quite easy to actually create a screenshot of the currently displayed screen and save it to the file system. If that's not the case this suggestion is moot, anyway.

For support purposes (for Hammerhead or for ki2 itself) it would be great if one could create a screenshot with the device itself. Taking a photo with a mobile phone during a ride is often not possible and if possible, most of the time the resulting picture quality is quite bad due to a bad angle, screen reflections, fingerprints and such. However, if one could configure the Di2 hood buttons (double press, long press) to immediately create a screenshot could really help here.

valterc, 2023-08-22:
> it is quite easy to actually create a screenshot of the currently displayed screen and save it to the file system

Probably doable for the ride app screen, not sure if we could get the screen outside the ride app or in the control panel for example.

kintrupf, 2023-08-23:
It would help a lot if it could be done for the ride app screen, all other screens are less important.
Thank you 😊

valterc, 2023-09-08:
Have a working version with this feature, still need to test in some rides.

Screenshot functionality is limited to the Ride App, screenshots are saved in `/mnt/sdcard/Pictures`.

closed 2023-09-09 by valterc

kintrupf, 2025-01-28:
Will this also be added to the newer versions using Karoo Extensions?

valterc, 2025-01-28:
Unfortunately this is not really possible with the new Karoo Extensions.

It is not possible for an App in android to capture the screen from a different App. It worked in the old Karoo SDK because Ki2 code was running inside the Ride App, so it could capture the Ride screen. Karoo extensions work in a very different way which does not allow for that.

PR #197 (2023-09-09): Add screenshot feature
Add feature to take screenshot during a ride via Di2 switches.

Closes #192