valterc/ki2 #195: when preload is activated 'ride resume' doesn't work good

Kraaien, 2023-09-02:
In an email conversation with H looking for a bug for the 'ride resume' function for the Karro2, the conclusion was as follows.
When 'preload' for the Ki2 app is activated, the 'ride resume' function doesn't work properly. I.e. when you stop for lunch and power-off the device and then after lunch you start the device again you don't get the possibility of 'ride resume'. You have to start a new ride. 
After the ride you then have two fitfiles. 

Can 'ride resume' in combination with 'preload' be fixed?

valterc, 2023-09-05:
Thanks for reporting. 

The preload is basically starting the ride app and then closing it, that action is likely making the ride app assume that you don't want to resume the previous ride and ignores it.

I can try to look into that, but I don't expect a fix from our side. The preload is already there as a workaround for the SDK data elements not showing, so from our end this is tricky to fix. If you are still in contact with H please ask them nicely to fix the original problem in the first place so that we don't need the preload 🙂

valterc, 2023-09-05:
Ok great 🙂 Thanks for the update.

valterc, 2024-12-28:
This will be fixed by the new version using Karoo Extensions.

closed 2024-12-28 by valterc