valterc/ki2 #203: Does the pin matter on Shimano eMTB?

uberamd, 2023-10-27:
I assume the answer to this is "no" but I want to make sure: does changing the default connectivity pin (the 6-digit number that defaults to 000000) impact Ki2 and its ability to connect to Shimano eMTBs?

For one reason or another the Ki2 app on my Karoo finds my eMTB, connects, but no data displays on both the Ki2 device list or within a profile with data fields (says no ebike is connected).

Just wanted to make sure I didn't set myself for failure right off the bat by changing the wireless pin. Unfortunately there's no easy way to reset it back to 000000 otherwise I'd have just tested that. Went through the code and I didn't see anywhere where a pin is transmitted during the connection so I think that's not the issue but asking anyway.

Thanks!

valterc, 2023-10-30:
Hi,

Ki2 connect to Di2 / STEPS via ANT, the connection PIN is only used for Bluetooth connections, so it's ok to change the PIN. Make sure ANT is enabled in Di2 settings.

> no data displays on the Ki2 device list

Could you explain what do you mean with no data display? Are you able to connect in Ki2? What is shown is the device list in Ki2 after you add the device?

> no data displays within a profile with data fields (says no ebike is connected)

Ki2 does not work in existing Karoo profile fields, you cannot use the Karoo fields. Instead you need to edit a profile and replace the Karoo fields with Ki2 fields.

See more details here: https://github.com/valterc/ki2#how-can-i-try-this-app
[How to Connect](https://user-images.githubusercontent.com/1299179/204136334-c8a5a395-c6b5-4d16-a8d5-ff1fa2dd726d.mp4)
[How to setup profiles with Ki2 elements](https://user-images.githubusercontent.com/1299179/204136325-69bcbf7a-b69f-45ed-96c4-3d9a52112089.mp4)

Also keep in mind support for STEPS / eMTB is limited to shifting information and battery level, there is no support for other information at this point.

uberamd, 2023-11-09:
Thanks for the reply! There was definitely a healthy amount of misunderstanding on my part. 

The good news is that I was able to get some of the Ki2 data fields to work! For example, eventually the Di2 gear shift count and gearing changes show up, and battery % did populate. 

Key word being eventually, sometimes I shift and it's reflected quickly, but most of the time the shift counts and gearing (front/rear etc) aren't updated as I change gears.

As for what I meant by no data display, it's this page on the Ki2 settings 

[image]

So it definitely is progress which I'm super happy about, this is great! However I'll need to dig in a bit to see why the data seems to be very delayed or dropping out causing shifts to not register, etc. Will look through existing bug reports for anything relevant. 

Thanks!

valterc, 2024-12-28:
Closing as this is stale.

closed 2024-12-28 by valterc