valterc/ki2 #211: ki2 apps appears to be causing disconnection issues for Garmin Varia Radar

alibikestuff, 2024-01-08:
Hi,

I notice in your ki2 apps page that your setup during testing included a Garmin Varia Radar, have you tested this on a Karro2 recently when running the ki2 app ?

The reason I ask is because on my last ride (yesterday), I received 8 separate Radar Disconnected errors and I am told by Hammerhead that they can see from the logs that the Ki2 app is causing major disturbance to the sensor connection ! :(

I have your ki2 app on my Karro2 and only have one bike with di2, the bike I rode on yesterday does not have di2, but I was finding that my HR from my watch wouldn't connect (Garmin Venu2) and the Radar didn't connect and start the light  when I started the ride, but when I turned the Radar light on manually, it worked but I had the 8 radar disconnect issues throughout the ride.

As the bike I ride on in the Winter does not have di2, is my only solution to deinstall the ki2 app to avoid these issues (as the ki2 app is continually looking for a di2 signal), or should it be sufficient to change the ki2 setting to NOT preload application data by default ? - as I don't have any ki2 data fields on the bike profile for this non di2 bike, I am assuming that it is the preload setting in the ki2 app that causes the disturbance ?

Thanks
Allison

valterc, 2024-01-08:
Hi, if you are still in contact with Hammerhead could you please ask for the logs? Would be nice to understand what is actually happening so we can fix it here 🙂

I haven't had problems in recent rides with with Garmin Varia and Garmin HR on bikes with Di2 and without Di2. I do get random Garmin Varia disconnects from time-to-time (like 1 or 2 every other week) but I also got those before using Ki2.

> The ki2 app is continually looking for a di2 signal

Ki2 will try to look for Di2 signal when you start or resume a ride, for the first 2/3 minutes. After that it becomes inactive.

I suggest you try to disable the Di2 device in Ki2: open Ki2, press on the Di2 on the list and hit the checkbox on `Enable`, so that the Di2 connection is disabled. Do that for all Di2 devices if you have more than 1. 

No need to change the preload setting or anything else because that part does not interfere with the ANT connections (the protocol used to communicate with Di2, Garmin Varia, etc.)

If you are able to correlate the disconnects with Ki2 being enabled, do try riding with Ki2 enabled/disabled a couple of times to see if it really correlates, please do reach out to see if we figure out why and fix the issue 🙂

robboyer, 2024-01-08:
FWIW I rarely have Varia disconnects while using Ki2, but they do occur.  Yesterday on a 58 mile ride I had it happen 2x.  But, my buddy who doesn’t have a Karoo but does have Varia also had a disconnect at exactly the same time of one of them. He normally has several per ride, like me.  He’s using a Wahoo.

Ki2 is working flawlessly for me.

alibikestuff, 2024-01-08:
> Hi, if you are still in contact with Hammerhead could you please ask for the logs? Would be nice to understand what is actually happening so we can fix it here 🙂
>
> I haven't had problems in recent [...]

alibikestuff, 2024-01-08:
Valter,

Thanks for your detailed reply.

I will try as suggested.

I will also see if Hammerhead can provide the logs.

I really think it might be some Garmin issue, but they exchanged my unit a couple of months ago when I explained that I had started to get several 'Radar Error' messages as well as multiple 'Radar Disconnected' on every ride !
It would be great to get to the bottom of this issue for sure. 

The 'Radar Error' issue is supposed to only occur when there is some kind of interference and it cannot get a signal to determine whether there is a vehicle, yet it is displaying vehicles correctly in the display. The radar disconnected issue shouldn't really be happening as the radar remains attached to my bike in the same place at the same short distance from my Karro2, both are annoying distracting messages causes a red banner text display to appear and a beep, on my rude yesterday this happened 8 times ! :(

I will keep you posted.

Thanks

Allison

alibikestuff, 2024-01-09:
Hi,

Further info. from Hammerhead team is:

Though I do not have the option to share the logs externally, I can help you with a screenshot of the event in the logs which is causing the sensor disconnection issues:

[image]

We were looking through Ki2 PRs and noticed this one https://github.com/valterc/ki2/pull/166/files which involves ki2 directly connecting to the ant plus plugin service and that it has mechanisms to kill the socket service and the plugin service.

I hope this helps you with the diagnostics.
Thanks.

alibikestuff, 2024-01-30:
Hi, I didn't hear any more on this, but since my last message, I have performed an update to my Garmin RTL515 Radar (an update from Garmin that appeared 'out of the blue' but states that one of the fixes is to 'reduce disconnect errors' !). Latest firmware on Garmin RTL515 is:
Light - 3.27
Radar - 3.34 (including fix to 'reduce disconnect errors')

I have also updated to your latest Ki2 software version 6.0 on my Karro2. I will see how the Radar behaves with the Karro2 this coming weekend.

valterc, 2024-01-31:
Thanks for the follow up 🙂 

From what I've seen from those Hammerhead logs and that comment about the Ki2 PR, I concluded that the problem was likely **not** related with Ki2. 

Ki2 is just using the ANT service normally, there are no attempt to reset the service or reset other connections. The problem is likely from the Karoo (software or hardware - might even be an ANT chipset/firmware problem) or indeed from the sensor itself (Garmin Varia), which after that firmware changelog, it appears to indeed be the case.

Hopefully it works well for you now!

closed 2024-05-27 by valterc