valterc/ki2 #214: SOC about 4% too low on ORBEA Rise 2024

RiseRider74, 2024-01-28:
The SOC ist not exactly the same as in Shimano Apps for my new ORBEA Rise M-Team (MY2024) with EP 801 RS and battery Orbea EBT360. While the Shimano App says 50% there are only 46% shown on my Karoo. These 4% remain constant when battery is getting lower. 

Just upgraded from 5.0 to 6.0 - same thing!

valterc, 2024-01-28:
Is it possible that the shimano app is showing a rounded (approximate) value? 50%, then 45% or 40% while the Di2 shows the real value?

Ki2 is just getting the value directly from the Di2 system, I'm not sure if there any offset or difference with an EP8 bike.

RiseRider74, 2024-01-28:
That is strange, of course the Shimano App shows exact values with 1% increments....