valterc/ki2 #215: screen turn on still not working....

proseLA, 2024-01-30:
first off, i thought i had updated my karoo software.  apparently i had not, i am still running 1.395.1557.12.  i will update prior to my next ride.

i am running 6.0 of your software.

i will make a comment on the UI.  under switch preferences, there is something called `Turn screen on`. i am not sure if that toggle switch is so clear as to when it is on or off.  i assume grey is on, white is off...  but i have tried it both ways.  and when the screen is off, hitting either button never turned the screen on.

with regards to functionality, i would think said toggle is unnecessary.  if someone hits either switch when the screen is off, why not just turn the screen on?  i can think of no instance where when the screen is off, and someone hits one of the buttons on the hoods, why they would not want the screen to automatically get turned on.  

but, i would love to hear why that might be the case.  i'm flexible.

in addition, when configuring said switches, i have also configured a single press on D-Fly Ch. 2 to Turn screen on.  unfortunately that did not work either.  i tried that with the aforementioned toggle set to grey as well as white.

so, i am not sure if i am doing something wrong with regards to my configuration on the ki2; or if this functionality is not working with the karoo firmware i have loaded.

i would like to hear how i should have it configured as well as if i should update my karoo to the latest firmware version.

but again, it is my opinion, that with regards to turning the screen on, if any switch is pressed, single, double or hold; and the screen is off, you should just turn the screen on, and then do the action configured.  

thanks again for this most awesome addition to the karoo.

#188

valterc, 2024-01-31:
Hi,

To clarify, the `Turn screen on` only works during a ride, when the ride app is active. It does not turn the screen on before starting a ride, while in the main menu, for example.

For the feature to be active the setting must be like this (it is active by default):
[image]

I need to check that feature after the last Karoo updates, I haven't used it recently, so it might have stopped working on some Karoo versions. Will update here.

proseLA, 2024-02-01:
i believe i tried this on my last ride and it did not work.  i will try again on my next ride; although much rain is currently forecast.  and yes, i tried it when the ride app is active; and that is the only time that i think important.

valterc, 2024-02-04:
Tried today an a ride, the feature is working.

What I did: installed Ki2 6.0, didn't change any setting, the `Turn screen on` feature is enabled like the image displayed on the previous message.

During the ride I enabled battery saver mode, waited until screen turned off and pressed a Di2 button associated with a D-Fly channel, in my case I used the hood buttons which are configured to move screen left/right on the Karoo. After pressing the Di2 button it takes around 1 sec and the screen turns on. The delay between pressing the Di2 button and screen turning on is because of Di2/ANT communication which is not immediate and also Android/Karoo screen which also doesn't respond immediately.

Summary:

Karoo 2 in battery saver mode, screen is off while riding --> press a Di2 button configured with a D-Fly channel (will not work when you press a button configured to shift gears) --> (1 sec delay) --> Karoo 2 screen turns on.

Tested multiple times. Please let me know if you are able to test.

proseLA, 2024-02-04:
i went for a ride yesterday.  the feature is still not working for me.

confirmed ki2 6.0. `turn screen on` is configured as the image above.

d-fly ch 1
single press -mark lap
double press - navigate left
hold - map/graph zoom out

d-fly ch 2
single press - turn screen on
double press - navigate right
hold - switch to map page

screen went into battery save mode while in a ride; pressing any of the button configurations did not turn the screen on.  after turning the screen on using one of the hard switches on the karoo devices, then the d-fly configurations work (except for the single press on ch2 as the screen is already on).

proseLA, 2024-03-19:
can anyone else confirm whether this is working or not?  @jeremywj?

i would prefer to close this issue if the problem only resides with me and my karoo.

so someone, please help us out and let us know if you find it working or not.

thanks.

kintrupf, 2024-08-15:
Same problem here, the screen of my Karoo 2 never turns on with ki2.
A long press of the left hood button should turn the screen on. Nothing happens.
A long press of the right hood button should sound the "bell", that's working fine. So no issue with the buttons itself.
I've also tried the option that any button press should turn on the screen. Does not work either.

I have ki2 7.1 installed and have set one of the Karoo profiles to battery safe mode, so that the screen is mostly off to save some battery life. Having the screen turn back on with a button press would be great!

proseLA, 2024-10-14:
ok, so as previously documented the barometer on my karoo 2 died.  and i now have a new karoo 2 with a working barometer.  and the screen turn on from the hoods now work.

based on this new data, i am going to say that there was some sort of issue with my old karoo 2 that prevented this functionality from functioning.  i am not sure if there was some other minor failing part on the old device or if that specific build of the karoo 2 had some sort of issue that prevented this functionality from working.

but i am going with some sort of problem with that specific karoo 2; and that there might be others out there with the same problem.

but given that my new device is now working; and more importantly without @valterc having a specific device to troubleshoot, i am closing this issue.  i think @valterc has bigger fish to fry... and to others who still have this problem, i feel your pain.

thanks again to @valterc for a killer plugin.

proseLA, 2024-11-09:
again, thanks to @valterc for providing this most USEFUL piece of software.  it is hard to imagine running the karoo without it.

ok i have some new information to report.  i am on the same device where the screen turn on from the hoods worked.  due to the problem that my karoo firmware version was not seeing my apple iphone bluetooth connection, i updated my karoo firmware to **1.516.1983.12**.  my Ki2 version is **8.1**.  and now the turn on from the hoods no longer works, although elements on the d-fly channels do work.

so, i am now of the opinion that it is either a software problem, or a configuration problem.

one of the interesting things is that on the [readme](https://github.com/valterc/ki2#:~:text=set%20the%20channels%20via%20the%20manufacturer%20official%20shifting%20app%20available%20on%20smartphones.%20), it says to  set the channels on the etube app.  well, other than setting the channel, i am not sure what else one can set there.  but when i look at what the ki2 is reporting, it says nothing is configured.  the see below images.

if anyone can help me out with regards to what i may be doing wrong; i would appreciate it.  i was **loving** being able to turn the screen on from the top of the 'hoods; and i would never have updated my karoo firmware; but the connection to my iphone is also quite important for me.

[image]
[image]
[image]

valterc, 2024-12-28:
This will be handled by the Karoo version using Karoo Extensions.

closed 2024-12-28 by valterc