valterc/ki2 #216: feature request: battery status graphic in gear display

robboyer, 2024-02-28:
Not sure it's possible, but It would be cool if the Ki2 graphic gear display data field could include a small battery icon or simple bar graph showing the Di2 battery charge level.  I know the charge level is available as separate data field, but this would save a field...and could turn red when getting low, etc.

valterc, 2024-03-13:
Hi Rob, that is a nice suggestion and totally possible. Let me sketch some designs and will come back to discuss.

robboyer, 2024-03-14:
awesome, thanks for considering it!

valterc, 2024-05-27:
Haven't had much time to work on this so far. Was thinking of what to change visually, probably the best is to follow HH design, this is what they have for gearing:

[image]

Will try to update the gearing fields with a similar design.

valterc, 2024-05-29:
Prototype:

[image]
[image]

Unfortunately we cannot make use of all the space like the default ones from hammerhead use.

valterc, 2024-05-29:
[image]
[image]
[image]

closed 2024-05-30 by valterc