valterc/ki2 #217: Beeps?

ddraber, 2024-04-13:
From the release notes it sounded as if the Di2 Interface could set a switch to "Beep" (_**Add switch press option to beep (use your Di2 hood switches + Karoo 2 as an annoying/bad horn!)**_). 

The K2 does not have a sound device, correct?  So how does this really work?

valterc, 2024-04-16:
K2 has a beeper/buzzer. It can beep before coming to a turn, when using navigation, or to alert on low battery, etc.

Ki2 uses the beeper for some events, like before a synchro shift or when the gear reaches the (lower/upper) limits. There is also an option in Ki2 to assign a beep/double beep to switch events. With that option, you can press a Di2 button, like the top hood switches, and the Ki2 will beep, you can even configure it to beep while you hold the Di2 switch down.

Of course it's not really a _horn_, but could be useful when cycling on shared path with people walking.

Let me know if that answers the question.

ddraber, 2024-04-16:
Thanks for the note.  I have been trying to get my K2 to make some sort of
sound since I first bought it.  I have all alerts on in all of my profiles
but nothing ever happens.  It probably is not your interface, but rather
something inside the K2 that keeps it quiet.

David

valterc, 2024-04-17:
That might be an hardware problem, double check the documentation from Hammerhead here: https://support.hammerhead.io/hc/en-us/articles/360056139654-Karoo-2-Audio-Alerts

If there's really an issue on your unit you might want to contact their support and get it replaced, for example on the community forums: https://support.hammerhead.io/hc/en-us/community/topics/360000095854-Tips-Troubleshooting

closed 2024-04-17 by valterc