valterc/ki2 #220: K3

CaesarrrZ, 2024-05-15:
Hi,

is there already a timeline or plan on supporting K3 with the Ki2?

Thanks so much
Paul!

goldenstarshine, 2024-05-16:
Hi, 

I second the request from @CaesarrrZ - that would be great!

Thank you for this great implementation so far!

Best regards,
g.

pictoruws, 2024-05-16:
+1 here

eloiRC, 2024-05-17:
I installed the apk on a new karoo and works but its imposible to pair with my ultegra

Nursedad, 2024-05-17:
Ditto. Got the current version sideloaded, but it won't find my electronic shifting groupsets produced by a Japanese company. :-)

heckcj, 2024-05-19:
Same issue as above.  Version 6 sideloaded to the new Karoo just fine, but it cannot find my Shimano Ultegra Di2 when searching.  Super bummed…

vmpg, 2024-05-20:
Hammerhead needs to activate the SDK on the K3, so I'll wait to buy it, who knows how long that will take.

heckcj, 2024-05-20:
I saw a post in the Karoo Users FB group from a K3 owner who sideloaded YouTube, Apple Music and Spotify apps.  He claimed all were working great… I’m not a software person, but wouldn’t these need the SDK to be activated if they are operating correctly?

If that’s not the case, maybe everyone on this email should reach out to Hammerhead Customer Support and request the SDK be activated.  I’m new to Di2 and was loving Ki2 on my Karoo 2.  I’m massively disappointed to find this functionality is not (currently) available with my new K3.

Chris Heck

vmpg, 2024-05-20:
Without the sdk the KI2 app does not work, you can start it but the di2 sensors are not found.
Other apps are not affected because they do not interact with the Karoo ride app.

heckcj, 2024-05-20:
Thanks - that makes sense.  I guess I will reach out to Hammerhead Customer Support and request the SDK be activated, noting that without it, I will return the K3 (I have 30 days) and just keep my K2 since an important piece of functionality I want and expected is not available.  I don’t know if that will do any good, but it’s the truth and worth a try.

Chris

CaesarrrZ, 2024-05-21:
> Ok, I sent support an email yesterday and filled out the form this morning.  So I’ve hit them up twice on this matter.
>
> Chris
>
>
>
>
> Yes, we all have to open a request on this regard to make them move. Go to this page<https://support.hammerhead.io/hc/en-us/requests/new> and write them a friendly message that the SDK availability is a decisive aspect in the purchase decision.

BRPaul

FlavourDave92, 2024-05-22:
There now is a [thread in the Hammerhead Support forum](https://support.hammerhead.io/hc/en-us/community/posts/25940221488539-Enable-SDK-for-Karoo-3) for this as well, feel free to express your support!

valterc, 2024-05-27:
Quick summary to keep everyone up to date:

To make Ki2 work on Karoo 3 we need to do two things:

1. Update ANT connectivity
2. Wait for Hammerhead to enable the SDK

We can easily fix the ANT connectivity without any support from Hammerhead - I'm working on this. As for the SDK this needs to be done by Hammerhead. I believe that this is a small fix/change from Hammerhead side. Let's wait a while but keep pressure on their support channels.

valterc, 2024-05-29:
ANT connectivity problems will be fixed by #225. This means we can pair/connect to Di2 from Ki2 in Karoo 3 - but without the SDK we cannot add any Ki2 data field to ride profiles or use any other functionality like audio alerts, etc.

heckcj, 2024-05-30:
Thanks for keeping us updated!  Will there be a new version of Ki2 forthcoming, or does this fix happen "behind the scenes?"

Chris

jpweytjens, 2024-06-10:
There's an official response from Hammerhead in the SDK [thread](https://support.hammerhead.io/hc/en-us/community/posts/25940221488539-Enable-SDK-for-Karoo-3?page=1#comments) where we're asking for an SDK for the last Karoo.

valterc, 2024-06-17:
> Thanks for keeping us updated! Will there be a new version of Ki2 forthcoming, or does this fix happen "behind the scenes?" 

The new [Ki2 7.0 release](https://github.com/valterc/ki2/releases/tag/7.0) has the new ANT connectivity update, which will support connections to Di2 on Karoo 3. 

However because Hammerhead still haven't enabled the SDK, it is not possible to display Ki2 data fields in the ride screen and most Ki2 features (sounds alerts, battery notification, etc.) will not work.

CaesarrrZ, 2024-06-17:
Very cool thanks! Will the control via Di2 buttons work already without the SDK?ThanksPaul

nguyenvo96, 2024-06-18:
@valterc thank you for updating Ki2, I was able to sucsssfully connect my 12spd di2. I understand that without SDK most features wouldn’t work. 

I was hoping that the overlay screens would work at least, but they do not show-up on the ride or any other screens. Does the overlay screens require SDK support? If not I would love to work with you get that working.

CaesarrrZ, 2024-07-25:
Hi!New update just came out.. SDK added? Nothing mentioned in the release notes..ThanksPaul

valterc, 2024-10-08:
It seems (from my exploration) that Hammerhead is developing 'Extensions' support for Karoo 3. I guess they are still developing it so things might change, for the time being it seems that Extensions can create 'sensors' and add elements to the ride screen.

Until they release something officially we'll have to wait (or buy a different unit).

jpweytjens, 2024-10-21:
Someone over at karoo.reddit.com contacted Hammeread support asking for an update on the SDK. Support replied that they're working on releasing an SDK, but didn't provide a timeline.

All details [here](https://www.reddit.com/r/Karoo/comments/1g8nlm8/k3_sdk_confirmed_but_no_timeline_yet/).

FlavourDave92, 2024-10-22:
> Someone over at karoo.reddit.com contacted Hammeread support asking for an update on the SDK. Support replied that they're working on releasing an SDK, but didn't provide a timeline.
> 
> All details [here](https://www.reddit.com/r/Karoo/comments/1g8nlm8/k3_sdk_confirmed_but_no_timeline_yet/).

yes, that someone is me. Jami from HH support responded very kindly. He even followed up with this:

>  Hi,
>  
> I know that several community members are waiting for the SDK! It'll take a little time to develop and prep it for release, but it will definitely open up lots of new potential when developers can create software for the new Karoo.
>  
> Don't hesitate to reach out if you need anything else!
>  
> Happy Riding,
> Jami
> Hammerhead
>

valterc, 2024-11-22:
We have some news from HH, extension support is officially available: https://github.com/hammerheadnav/karoo-ext

It appears to be quite complete in terms of features, but I haven't used it yet. It's going to take some time to port Ki2, but looking at the documentation it looks like it will support all Ki2 features.

robboyer, 2024-11-23:
Great news!  Thanks for letting us know and for taking this on.  

Coffee forthcoming….

vdubgeek, 2024-11-23:
Appears that the extensions have been officially released, https://github.com/hammerheadnav/karoo-ext .

bigfloor, 2024-11-23:
Really hope that's enough to get Ki2 on the new Karoo (it doesn't mention there that it's an SDK)

heckcj, 2024-11-27:
Same question as posed by dylanbeechey… Hammerhead just released a software update for the Karoo 2 and Karoo (3). According to the release notes, “New Extensions are supported on both the new Karoo and Karoo 2. Previous apps built on the Karoo 2 SDK will not work with the new Extensions SDK.”  Does this mean this update will break Ki2 on a Karoo 2?

dylanbeechey, 2024-11-27:
@heckcj check out @valterc 's response here: https://github.com/valterc/ki2/issues/253 Looks like K2 should still be good and he's working on K3 🙏🏽🥳

heckcj, 2024-11-27:
Perfect, and thanks for the follow-up info!

valterc, 2024-12-03:
Hey, quick update on the support for Karoo 3. There are some quirks with the Karoo Extensions (the new SDK), that Hammerhead is working to fix, hopefully soon. 

Until then, I'm trying to understand the best way to move to extensions. It's likely that Ki2 will lose some functionality:
- Will remove some of the button commands, for example the screenshot (is anyone using that anyway?)
- The ride overlay will change to a native android overlay - not sure how good it will be
- Still unclear if we will have support for storing the gear information in the FIT file
- Still unclear if we can have all the existing graphical ride elements

It's going to take a few weeks until we have something good enough to test.

pencerw, 2024-12-03:
@valterc thanks so much for working on this, and for the update!

I'm sure everyone has their personal preferences, but personally I'd love if I could still a) display what gear I'm in on my K3, and b) use the customizable buttons on my GRX shifters to control things like map zoom on the K3. 

...but any integration is better than no integration, and whatever Ki2 ends up looking like I appreciate your efforts!

dylanbeechey, 2024-12-04:
Thanks so much @valterc ! Getting any type of integration back and being able to see gears would be a win for me. Appreciate what you do!

bigfloor, 2024-12-12:
@valterc Bought you a beer for all your work !!

Javiuchi, 2024-12-17:
I sent you another beer. Thank you for your excellent work.

syntactosemantic-error, 2024-12-23:
Thanks for the updates @valterc. Bought you coffee. I've never used an github app that works this good. Hope there is an update for K3 too.

klausmueller1983, 2024-12-27:
hello, is there already a new timeline as to when KI2 will be available on the current Karoo?

valterc, 2024-12-27:
Hi everyone, thanks for support, I have some news to share about Ki2.

I have a fully working version for the new Karoo extensions, that version should be compatible with Karoo 3 (I haven't tested). 

There are a few notes about this version that I will share for potential feedback from you:

- The plan moving forward is to mainly use the native Karoo profile elements, essentially we will add the Di2 as a new device to Karoo and then use the Gears elements from Karoo itself. (FIT recording will be supported)
- There are still some bugs/limitation that Hammerhead are working on their end.
- The new version of the app will not be backwards compatible. This means anyone using Ki2 will have to re-install the app.

I will soon release an update to Ki2 that will disable future auto-updates for the current version. Likely next week I can release a first version with the Karoo extension and support for Karoo 3.

dylanbeechey, 2024-12-27:
Awesome Valter! Thanks so much for the hard work on this. Can't wait to start using Ki2 again!

Gekran, 2024-12-28:
Great news at the end of this year Valter. I am exited to use the app Ki2 back again. Thanks for your outstanding work on this.

FlavourDave92, 2024-12-28:
Awesome!! Thank you, Valter!

Tauhax, 2024-12-28:
Great job Valter 🙌

junyfly, 2024-12-29:
great jod ！n thanks @valterc

95wildtt, 2025-01-06:
The New 10.0 version is working perfect on my K2... Thanks Valter, sent you a bunch of coffee :-)

valterc, 2025-01-06:
Finally closing this issue, as Ki2 is now available for Karoo 3. First release compatible with Karoo 3 here: https://github.com/valterc/ki2/releases/tag/10.0

closed 2025-01-06 by valterc