valterc/ki2 #221: Problem with K3 not finding DI2 in KI2 APK file

Marc3773, 2024-05-21:
Dear,
Can somebody help me with the problem that after intalling ki2 on my new hammer head karoo 3 device di2 is not found.
with my former models the karoo 1 and 2 there was no problem.
I Hope somebody can help me or tell this will be solved in a new version of Ki2?

I am looking forward to the response.

Best Regards,
MArc

FlavourDave92, 2024-05-21:
Hi MArc, 

it seems like Hammerhead still has to enable the SDK for the karoo 3 like they had for 1 and 2 before (see [this thread](https://github.com/valterc/ki2/issues/220)).

You can go to [this page](https://support.hammerhead.io/hc/en-us/requests/new) and write Hammerhead a friendly message that the SDK availability (and thereby ability to use di2 with ki2) is a decisive aspect in your purchase decision.

Cheers

FlavourDave92, 2024-05-21:
Oh dear... that doesn't sound good...

heckcj, 2024-05-22:
I sent an email to Hammerhead Support about this same matter and got a similar (although less specific) response:

"As of now, we have not implemented SDK capabilities on new Karoo and due to this, you wouldn't be able to install any third-party apps on new Karoo.
 
Though we do not have a timeline to share with you, we hope to see this SDK capability in the future.
 
Your request will be forwarded to the product team."
 
I have informed them I will be taking advantage of the 30-day return policy (and simply continue to use my K2) is SDK is not activated in the next few weeks.

FlavourDave92, 2024-05-22:
Same here, I think we need more people to write them that the SDK for Ki2 is crucial:

> As of now, we have not implemented SDK capabilities on new Karoo and due to this, you wouldn't be able to install any third-party apps on new Karoo.
>  
> Though we do not have a timeline to share with you, we hope to see this SDK capability in the future.
>  
> Your request will be forwarded to the product team. 
>  
> Let us know if you have any further queries.
>

FlavourDave92, 2024-05-22:
There now is a [thread in the Hammerhead Support forum](https://support.hammerhead.io/hc/en-us/community/posts/25940221488539-Enable-SDK-for-Karoo-3) for this as well, feel free to express your support!

valterc, 2024-05-27:
Thanks for the input and discussion here, looking forward to have Ki2 on the Karoo 3. I will close this issue and let's keep the discussion and tracking this on the #220.

closed 2024-05-27 by valterc