valterc/ki2 #228: Ki2 for K3

gdouglas32, 2024-06-17:
I cannot get this work on the K3, keeping getting the following error "Ki2 keeps stopping" and will not open to the dashboard.

valterc, 2024-06-17:
Karoo 3 is not fully supported because of the lack of SDK on Karoo 3, but the app should still open. If you haven't done it, please restart Karoo and try again. 

If it still doesn't work it would be nice if you could provide the logs, as I don't have a K3 to test.

How to get the logs:
- Install Ki2 normally
- Use ADB (the app that you've used to install Ki2) and run:
- `.\adb.exe shell` (press Enter)
- Inside the ADB shell run:
- `logcat | grep ki2` (press Enter)
- Now open Ki2 in Karoo, you should get some message on the terminal, take a screenshot or copy the text to here
- Press Cmd/Ctrl-C on the terminal and close the app on K3
- Still inside the ADB shell run:
- `logcat *:E` (press Enter)
- Now open Ki2 in Karoo, you should get some message on the terminal, take a screenshot or copy the text to here

nguyenvo96, 2024-06-18:
Restarting the K3 allowed me to proceed to the pairing screen, and I was successfully able to connect my ultegra 12spd di2.

dylanbeechey, 2024-07-06:
I was able to install it on the new Karoo without issue and it appears to connect based on the connection status and reading the correct gears, however I'm not getting any shifting information on a ride although I've customized shifter buttons, they don't seem to be saving correctly or are not working. Haven't tried to re-install the app yet which is probably what I'll do next. 

Was using the app on my K2 previously and love it. Thank you @valterc !

valterc, 2024-07-06:
Thanks for testing @dylanbeechey ! No need to re-install the app, while you will be able to connect to Di2, the shifting information and shifter buttons, etc. won't work because of the lack of Karoo SDK in Karoo 3. 

Hammerhead didn't enable SDK on the new Karoo 3, so Ki2 will not work as it normally would in Karoo 2.

More details here: https://support.hammerhead.io/hc/en-us/community/posts/25940221488539-Enable-SDK-for-Karoo-3

I'm closing this issue as duplicated because we already had another one related to this topic at #220.

closed 2024-07-06 by valterc