valterc/ki2 #233: App closing when trying to connect to Di2

Foramen1, 2024-08-08:
Sideloaded the 7.0 version to my Karoo2.0.
I can open the app but the moment I try to connect to my Di2 and press the "+" it will stop working.
Reboot did not solve the issue.

alibikestuff, 2024-08-27:
I have had app closing issues each time I start up my Karro2, but the last 2 rides, have also had issues with resuming rides - weird data, including a wildly fluctuating average speed - 144mph is a bit extreme !

I contacted Hammerhead and they say they have seen this issue before and it has been where ki2 is installed.

I am reluctantly going to deinstall ki2 and see how it goes.

pictoruws, 2024-09-03:
Thank you for building this great app for the Karoo2

I have had to stop using Ki2 because of constant crashes of the app with HX errors every 30secs with the screen freezing and losing ride data.  Never happened before.  How get I get you any logs to help troubleshoot.  Note this issue went away once I disabled Ki2.  I love your work with this capability and hope it is not permanently broken.  Please advise on anything I can do to get this resolved.

valterc, 2024-09-03:
Hi all, 

I have only seen the HX error once and didn't manage to capture the logs, so I will need your help to diagnose and fix those problems.

Here is what I need you to do:

- Make sure you have [Android Platform Tools](https://developer.android.com/tools/releases/platform-tools#downloads) in your computer (it's the same tool that was used to install Ki2)
- Start your Karoo and use it normally, and do the actions that will cause errors
  - If the problem is when Ki2 is adding a new Di2 sensor, then do that
  - If the problem if when using the ride app, then start a ride
  - Make sure you see the errors in Karoo
- Without shutting down the Karoo, connect it to your computer
- In a Terminal/Powershell window, in the directory of the extracted Android Platform Tools, run each of the following commands separately:
  - `.\adb.exe logcat -d -e ki2`
  - `.\adb.exe logcat *:E -d`
- Copy the output of those commands here on github

If the output is too large to paste on github, please use a service like [pastebin](https://pastebin.com/).

Let me know if that makes sense or if I can provide more guidance 🙂

pictoruws, 2024-09-03:
Makes sense. Let me give it a shot tonight. Stand by please.   Thank you.

Foramen1, 2024-09-04:
PS C:\Users\Jasper\Downloads\platform-tools> .\adb.exe logcat -d -e ki2
* daemon not running; starting now at tcp:5037
* daemon started successfully
- waiting for device -
--------- beginning of system
09-04 14:44:25.874   710  1880 I ActivityManager: Start proc 3023:com.valterc.ki2/u0a60 for broadcast com.valterc.ki2/.karoo.Ki2BroadcastReceiver
09-04 14:48:51.983   710   723 I ActivityManager: START u0 {act=android.intent.action.MAIN cat=[android.intent.category.LAUNCHER] flg=0x10000000 pkg=com.valterc.ki2 cmp=com.valterc.ki2/.activities.main.MainActivity bnds=[38,518][232,642] (has extras)} from uid 10038
09-04 14:48:54.554   710   723 I ActivityManager: START u0 {cmp=com.valterc.ki2/.activities.devices.add.AddDeviceActivity} from uid 10060
--------- beginning of crash
09-04 14:48:58.826  3023  3124 E AndroidRuntime: FATAL EXCEPTION: ServiceHandler
09-04 14:48:58.826  3023  3124 E AndroidRuntime: Process: com.valterc.ki2, PID: 3023
09-04 14:48:58.826  3023  3124 E AndroidRuntime: java.lang.RuntimeException: com.dsi.ant.channel.ChannelNotAvailableException: Could not acquire channel. Reason = SERVICE_INITIALIZING
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.handler.ServiceHandler$1.accept(ServiceHandler.java:60)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.handler.ServiceHandler$1.lambda$accept$0$com-valterc-ki2-services-handler-ServiceHandler$1(ServiceHandler.java:58)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.handler.ServiceHandler$1$$ExternalSyntheticLambda0.run(D8$$SyntheticClass:0)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.Handler.handleCallback(Handler.java:790)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:99)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:164)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.handler.ServiceHandler.lambda$new$0$com-valterc-ki2-services-handler-ServiceHandler(ServiceHandler.java:29)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.handler.ServiceHandler$$ExternalSyntheticLambda1.run(D8$$SyntheticClass:0)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at java.lang.Thread.run(Thread.java:764)
09-04 14:48:58.826  3023  3124 E AndroidRuntime: Caused by: com.dsi.ant.channel.ChannelNotAvailableException: Could not acquire channel. Reason = SERVICE_INITIALIZING
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.dsi.ant.channel.ChannelNotAvailableException$1.createFromParcel(ChannelNotAvailableException.java:83)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.dsi.ant.channel.ChannelNotAvailableException$1.createFromParcel(ChannelNotAvailableException.java:79)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.Parcel.readParcelable(Parcel.java:2860)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.Parcel.readValue(Parcel.java:2754)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.Parcel.readArrayMapInternal(Parcel.java:3123)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.BaseBundle.initializeFromParcelLocked(BaseBundle.java:273)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.BaseBundle.unparcel(BaseBundle.java:226)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at android.os.Bundle.getParcelable(Bundle.java:939)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.dsi.ant.channel.AntChannelProvider.acquireChannelOnPrivateNetwork(AntChannelProvider.java:341)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.dsi.ant.channel.AntChannelProvider.acquireChannelOnPrivateNetwork(AntChannelProvider.java:288)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.ant.AntManager.getScanAntChannel(AntManager.java:244)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.ant.scanner.AntScanner.startScan(AntScanner.java:32)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.Ki2Service.processScan(Ki2Service.java:580)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.Ki2Service.access$2300(Ki2Service.java:79)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.Ki2Service$1.lambda$registerScanListener$19(Ki2Service.java:318)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.Ki2Service$1$$ExternalSyntheticLambda10.run(D8$$SyntheticClass:0)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        at com.valterc.ki2.services.handler.ServiceHandler$1.accept(ServiceHandler.java:55)
09-04 14:48:58.826  3023  3124 E AndroidRuntime:        ... 8 more
09-04 14:48:58.833   710   722 W ActivityManager:   Force finishing activity com.valterc.ki2/.activities.devices.add.AddDeviceActivity
09-04 14:48:58.854   710   733 I ActivityManager: Showing crash dialog for package com.valterc.ki2 u0
09-04 14:49:02.150   710   722 I ActivityManager: Killing 3023:com.valterc.ki2/u0a60 (adj 0): crash
09-04 14:49:02.152   710   722 W ActivityManager: Scheduling restart of crashed service com.valterc.ki2/.services.Ki2Service in 1000ms
[... cut ...]
09-04 14:50:41.016   589   649 E XTCC-6.1.2.10-a: [FDAL_OSListener] handleLocationUpdate:  failed: 2
09-04 14:50:42.011   589   649 E XTCC-6.1.2.10-a: [FDAL_OSListener] handleLocationUpdate:  failed: 2
09-04 14:50:43.011   589   649 E XTCC-6.1.2.10-a: [FDAL_OSListener] handleLocationUpdate:  failed: 2
09-04 14:50:46.012   589   649 E XTCC-6.1.2.10-a: [FDAL_OSListener] handleLocationUpdate:  failed: 2
09-04 14:50:47.014   589   649 E XTCC-6.1.2.10-a: [FDAL_OSListener] handleLocationUpdate:  failed: 2

Foramen1, 2024-09-04:
Let me know if this is helpful.
I logged the Karoo, crashed when trying to add a device to Ki2.
Did this 3 times over. 

https://pastebin.com/9vEtYjpR

valterc, 2024-09-04:
Thanks @Foramen1, that is very useful indeed. I can see the errors and their source, but I'm still a bit unclear why they are happening.

Essentially the error happens because the ANT service is not yet initialized, now why is the ANT service not initialized?... I will investigate some more on my end. 

@pictoruws if you could also provide your logs, that would be helpful 🙂

mabar81, 2024-09-05:
Hello @valterc , I have the same issue starting from the past week.

From my logs I have this stacktrace:

09-05 11:53:10.885  3990  3990 E AndroidRuntime: Process: io.hammerhead.activityservice, PID: 3990
09-05 11:53:10.885  3990  3990 E AndroidRuntime: java.lang.IllegalStateException: This app has been built with an incorrect configuration. Please configure your build for VectorDrawableCompat.
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at androidx.appcompat.widget.ResourceManagerInternal.checkVectorDrawableSetup(ResourceManagerInternal.java:31)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at androidx.appcompat.widget.ResourceManagerInternal.getDrawable(ResourceManagerInternal.java:2)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at androidx.appcompat.widget.ResourceManagerInternal.getDrawable(ResourceManagerInternal.java:1)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at androidx.appcompat.content.res.AppCompatResources.getDrawable(AppCompatResources.java:5)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at com.valterc.ki2.karoo.battery.LowBatteryNotification.showLowBatteryNotification(LowBatteryNotification.java:31)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at com.valterc.ki2.karoo.battery.LowBatteryHandler.lambda$notify$3$com-valterc-ki2-karoo-battery-LowBatteryHandler(LowBatteryHandler.java:147)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at com.valterc.ki2.karoo.battery.LowBatteryHandler$$ExternalSyntheticLambda7.run(D8$$SyntheticClass:0)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at android.os.Handler.handleCallback(Handler.java:790)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at android.os.Handler.dispatchMessage(Handler.java:99)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at android.os.Looper.loop(Looper.java:164)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at android.app.ActivityThread.main(ActivityThread.java:6518)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at java.lang.reflect.Method.invoke(Native Method)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:438)
09-05 11:53:10.885  3990  3990 E AndroidRuntime:        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:807)

I hope it helps

mabar81, 2024-09-05:
Hi again @valterc , after writing I've checked mi di2 battery and it was at 15% (so below the 20% warning threshold). I've charged it and now everything works fine again.
Even from my stacktrace it seemed related to the battery, maybe something with the battery notification events...

pictoruws, 2024-09-06:
That battery issue sounds similar to what happened with me.  I tested the Karoo on another di2 bike with a charged battery and no errors.

Foramen1, 2024-09-06:
@valterc I'm just reading the other replies regarding the battery of the Di2, I discovered later on that I got an earlier Di2 setup without the bluetooth built-in. 
Wanted to test if the app worked before I would purchase the €80 D-fly module.
Considering the other messages regarding the battery percentage, this may influence the app as well?
As it can not find any device?

awidegreen, 2024-09-06:
> That battery issue sounds similar to what happened with me. I tested the Karoo on another di2 bike with a charged battery and no errors.

Same here; charging the battery solved the crash for me as well (battery was at 15% when the app was crashing). Note: I also tested downgrading the app to v6.0 and got the same problem.

valterc, 2024-09-14:
I just released Ki2 version [8.0](https://github.com/valterc/ki2/releases/tag/8.0), please update Ki2 directly from Karoo or via your computer. The problems should be fixed, let me know if you still have problems.

closed 2024-09-14 by valterc