valterc/ki2 #234: Zoom in/out sometimes "overshoots" and zooms completely in or out

kintrupf, 2024-08-15:
On my di2 I use a double press of the left and right hood button to zoom in or out of the map. 
Most of the time this works perfectly fine, but once in a while ki2 seems to register multiple button presses/commands for one double press and zooms all the way in or out of the map. Of course this makes the map instantly useless, until a reverse zoom is manually applied multiple times 😉

Could you restrict zoom in/out to exactly _one step_ and ignore all other zoom commands that might be in a command queue (if there's one) or are received within the next half second of the first command?

If only a single zoom command is executed each time the long press of a button could also be used to zoom the map.
Right now the lag of the Di2 buttons makes this almost impossible. Either the long press is too short, and no zoom is executed, or the press is slightly too long and multiple zooms (mostly too much) are executed.

I use ki2 7.1, but the problem already exists for a very, very long time.

valterc, 2024-08-31:
This is indeed a long standing issue, the original HH di2 support had the same issue, because the underlying problem is in Di2.

I will add two other settings:
- Hold to zoom in/out only once
- Hold to zoom in/out slower, so that you can let go of the Di2 button on desired zoom

valterc, 2024-08-31:
Ended up improving the current zoom setting, so the zoom will work much better without the need to change any setting. 

Still, added the two other settings (as described above) for more customization.

Will be available in the next Ki2 update.

closed 2024-08-31 by valterc

valterc, 2024-09-14:
I just released Ki2 version [8.0](https://github.com/valterc/ki2/releases/tag/8.0), please update Ki2 directly from Karoo or via your computer. The problem should be fixed, let me know if you still have problems.

kintrupf, 2024-09-23:
The two new options work very well for me, thank you!

However, the "old" option still produces this problem occasionally when used with a double-press.
I did a longer ride with the new Ki2 version and three times the zoom in/out did not stop after a single double-press.

PR #236 (2024-08-31): Improve map zoom when holding switch
Fixes #234