valterc/ki2 #240: how to downgrade?

proseLA, 2024-09-14:
i side-loaded the latest version (8) and went for a ride today.

curiously enough my temperature sensor and my vertical ascent sensor stopped working at exactly the same time.

as i want to rule out a problem with ki2, i attempted to downgrade back to 7.1.  i tried the following 2 commands:
```
$ ./adb install -r -d ki2-7.1-release.apk
$ ./adb install -r ki2-7.1-release.apk
``` 
but in both situations i received:

```
Performing Streamed Install
adb: failed to install ki2-7.1-release.apk: Failure [INSTALL_FAILED_VERSION_DOWNGRADE]
```
any ideas as to how to uninstall or successfully downgrade the software?

i would like to rule out ki2 as the cause of my sensor problem.

thanks in advance.

valterc, 2024-09-14:
Hi, Android does not allow downgrades, you will need to uninstall Ki2 first and then install an older version. 

What seems to be the problem with your other sensor?

proseLA, 2024-09-14:
my karoo is no longer seeing the temperature or any vertical information.  just very curious it started after i loaded ki2 version 8.
[image]

closed 2024-09-20 by valterc