valterc/ki2 #247: EP8 Support?

sfsanchez, 2024-10-13:
Hi!

ki2 is great - thank you for creating this app! I've been using it on my Karoo 2 to connect the di2 units on my road and gravel bikes and it works great; however, I've never been able to get it to work properly with the EP8 unit on my 2022 Orbea Rise H15. I've read other threads where it has worked for people, so not sure why I can't get it to work properly. I'm running the latest firmware on all units - Karoo 2, EP8, and latest version of ki2. 

I can add the EP8 and connect to it (with good signal); however, it never loads data from the EP8 (just spinning wheels under the device details section). Eventually, one section will populate with the EP8 device details (serial number, software version, etc.), but the data never populates. I've have added the various data fields to the bike profile (including "SHF Bat %"), but the data never appears. 

Any ideas how I can fix this? I'm only looking to add the bike's battery percentage to my bike profile. 

Thank you!

Cheers,
Scott

sfsanchez, 2024-10-14:
Sorry. It's working now (AWESOME!). Not sure what happened - I let it sit overnight and now all is good. 

Thank you!

closed 2024-10-14 by sfsanchez

valterc, 2024-10-15:
Hey, I didn't have a chance to reply, good that you solved the problem! 🙂 The di2/bike needs to be active to broadcast data.

Keep in mind that some features from EP8/Steps will not be available, for example data such as power and speed. Those fields should be available in the broadcasted data but we need to understand and obtain them. I don't have an EP8 bike so I cannot add those, if you are a developer feel free to help with that.

sfsanchez, 2024-10-15:
Hi Valter,Thank you so much for the quick response. Yeah, it's working perfectly now. For the EP8, I only need it to show the batter % - so this is perfect. Unfortunately, I'm not a developer, so I don't have anything that I can offer other than saying thank you for creating this app! I've been following the SDK issue with the Karoo3. Hope fully, they will release it soon so your app can be available on the Karoo3. I definitely won't be getting the new unit until they make this change. Cheers,Scott